roleoroleo/yi-hack-Allwinner #79: FTP Upload config will not be saved in the Web-UI

JakobStadlhuber, 2020-07-04:
If I change the configuration in the FTP Upload section it will be not saved.
Now I want to edit it manually, but how will the password be hashed, plain password is not working.

Version: 0.1.7

roleoroleo, 2020-07-04:
Try to clean the cache of your browser.

JakobStadlhuber, 2020-07-04:
Thanks fixed it, is it possible to hash the FTP password in the future? At the moment it is saved as a plain pw.
Second question: when will the files uploaded after a motion detection? Is there a log to see if something failes?

Catfriend1, 2020-07-12:
FTP upload is happening every 45 seconds after the last upload execution. Failed items are retried 45 seconds later.

It can be changed by editing ftppush.sh
> SLEEP_CYCLE_SECONDS="45"

I chose this interval because my ftp server , a debian machine, uses a 60 sec interval to check for new uploads and forwards them to the tool dvr-scan (which can filter out false alarms) .

Catfriend1, 2020-07-12:
The log is here: /tmp/ftppush.log

closed 2020-11-21 by roleoroleo