roleoroleo/yi-hack-Allwinner #81: Yi-Hack V0.1.7 no rtsp

murigny, 2020-07-05:
Hello
v0.1.7 success installed on 2 cams
rtsp (high) work after install but, if you reboot, no more stream.

Catfriend1, 2020-07-12:
I have this problem too after upgrading from v0.1.6 to v0.1.7

Catfriend1, 2020-07-12:
Could it be commit https://github.com/roleoroleo/yi-hack-Allwinner/commit/9ebd1173a24aa51e4d7a0a20e5ddac3ca4ae1e23 to blame here?

roleoroleo, 2020-07-12:
@Catfriend1 
Did you solve?

Catfriend1, 2020-07-12:
No, didn't compile it.

murigny, 2020-07-12:
Trying all configuration combination and nothing on high res.
Try with motion, ispy and vlc
When i use vlc, there is something on stream (timeline work) and codec info is H264 but no picture.
Sometime it work for a few minutes and then down

roleoroleo, 2020-07-12:
The commit you mentioned was overwritten.
The RTSP server is the same between version 0.1.6 and 0.1.7.

murigny, 2020-07-12:
I'm trying to recompile, same result

roleoroleo, 2020-07-12:
Yes, thank you.

murigny, 2020-07-13:
Hello
I'm looking about logs but there is nothing strange in it.
Is there any way to add dmesg command in futur release ?

roleoroleo, 2020-07-13:
With 0.1.6, does it work?

murigny, 2020-07-13:
Yes, work with 0.1.6. some artefacts but work

Catfriend1, 2020-07-13:
0.1.6 was the last version it worked for me, too.

AymericM78, 2020-07-13:
Hello folks!

Same issue here, and it's weird because I have 2 cams, 2 Yi 1080p Home bought the same day.
The first one has the following configuration : 
- Firmware Version | 0.1.7
- Base Version | 8.2.0.0A_201912270941
- Model Suffix | y20ga
- Serial Number | 9FUSY31N9ZHVJ7200104

And RTSP is working like a charm.

The second one has the following configuration : 
- Firmware Version | 0.1.6
- Base Version | 8.2.0.0A_201912270941
- Model Suffix | y20gam
- Serial Number | 9FUSY31MVL3J2C200104

And RTSP is responding on VLC without video, audio only.
I was on 0.1.6 and it was working.
I have upgraded to 0.1.7 and the RTSP was KO.
Then I have removed hack (with https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack) and then reinstall 0.1.6.
But the issue still there.

I hope it will help to troubleshoot and found the cause and resolution.

Thanks!

murigny, 2020-07-13:
try compiling without "Move rmm start position" commit: same issue

murigny, 2020-07-13:
> Then I have removed hack (with https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack) and then reinstall 0.1.6.
> But the issue still there.
> 
> I hope it will help to troubleshoot and found the cause and resolution.
> 
> Thanks!

same here.
try removed hack + reset cam + flash 0.1.6 = issue still here.
something is broken. :(

roleoroleo, 2020-07-13:
If you restore an older version you have to remove the comment sign '#' before rmm inside init.sh.

roleoroleo, 2020-07-13:
I will investigate it.

murigny, 2020-07-13:
thanks, i use low res 👍 
it's a strange issue

roleoroleo, 2020-07-13:
Could you try with and without audio?

roleoroleo, 2020-07-13:
I don't understand.

- I removed the hack.
- I checked that the hack was correctly removed.
- I reinstalled 0.1.7.
- All is working properly: low and high res.

roleoroleo, 2020-07-13:
Or better...
Dump your flash and share it with me.
See the wiki.
mtdblock2 and mtdblock3 are enough.

murigny, 2020-07-14:
here is backup of 0.1.7 with rtsp high broken
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/4917928/backup.zip)

roleoroleo, 2020-07-14:
Ok, thanks.

roleoroleo, 2020-07-14:
Very strange, in your backup you miss init.sh file.
How does it work?

roleoroleo, 2020-07-14:
Used your file system and your config.
All is working properly.
What about a network issue?

murigny, 2020-07-14:
cams are on two different AP (for proximity) on wifi (N)
first at 6meters
second at 4 meters

murigny, 2020-07-14:
This is VLC media info when streaming rtsp high (no picture but timeline work)

[image]

roleoroleo, 2020-07-14:
Ok, are you able to do some tests?
If yes, try this please.

Restore the rmm start in file /home/app/init.sh
From:
```
...
#sleep 2
#./rmm &
#sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./watch_process &
/usr/sbin/telnetd &
/home/yi-hack/script/system.sh
```
To:
```
...
sleep 2
./rmm &
#sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./watch_process &
/usr/sbin/telnetd &
/home/yi-hack/script/system.sh
```

Change the middle part of /home/yi-hack/script/system.sh replacing all if [[ $(get_config DISABLE_CLOUD)
To:
```
if [[ $(get_config DISABLE_CLOUD) == "no" ]] ; then
    (
        cd /home/app
        if [[ $(get_config RTSP_AUDIO) != "none" ]] ; then
            killall rmm
            sleep 2
            OLD_LD_LIBRARY_PATH=$LD_LIBRARY_PATH
            export LD_LIBRARY_PATH="/home/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib"
            ./rmm &
            export LD_LIBRARY_PATH=$OLD_LD_LIBRARY_PATH
            sleep 4
            dd if=/tmp/audio_fifo of=/dev/null bs=1 count=8192
        fi
		sleep 2
        ./mp4record &
        ./cloud &
        ./p2p_tnp &
        ./oss &
        if [ -f ./oss_fast ]; then
            ./oss_fast &
        fi
        if [ -f ./oss_lapse ]; then
            ./oss_lapse &
        fi
        ./watch_process &
    )
else
    (
        cd /home/app
        if [[ $(get_config RTSP_AUDIO) != "none" ]] ; then
            killall rmm
            sleep 2
            OLD_LD_LIBRARY_PATH=$LD_LIBRARY_PATH
            export LD_LIBRARY_PATH="/home/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib"
            ./rmm &
            export LD_LIBRARY_PATH=$OLD_LD_LIBRARY_PATH
            sleep 4
            dd if=/tmp/audio_fifo of=/dev/null bs=1 count=8192
        fi
		sleep 2
        # Trick to start circular buffer filling
        start_buffer
        if [[ $(get_config REC_WITHOUT_CLOUD) == "yes" ]] ; then
            ./mp4record &
        fi
    )
fi
```

Try with and without audio.

murigny, 2020-07-14:
try your tips:
nothing change, with or without audio.
And what about "ARCHIVE=live.2020.05.15.tar.gz" perhaps this update is buggy ?

roleoroleo, 2020-07-14:
Ok, another step.
Try to stop manually rRTSPServer and run h264grabber with debug option.
Just to see if there is something bad.

murigny, 2020-07-14:
nothing strange:
- many "frame detected - frame_res: 1080 - frame_len: 2849 - frame_counter: 2417" len change and counter progress
- "SPS detected - frame_res: 1080 - frame_len: 20 - frame_counter: 2408" regularly
- "frames lost: 2" sometime

roleoroleo, 2020-07-14:
?!?
Clean the SD and try again to load 0.1.6.

murigny, 2020-07-14:
nothing. orange led, no reboot, no "clic".
sorry, cam is dead

AymericM78, 2020-07-14:
@murigny I had this yesterday.

Retry the "Remove hack" procedure: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack
then reset and pray

murigny, 2020-07-14:
@AymericM78 : i've tried this method with no success.

cam is going to litter

roleoroleo, 2020-07-14:
Connect the serial cable and access to uboot.
We will restore the cam.

murigny, 2020-07-14:
Sorry roleoroleo, i give up.
Thanks for your involvement ;)

roleoroleo, 2020-07-14:
I'm sorry.
If you want to try in the future, I will publish a wiki to unbrick the cam.

roleoroleo, 2020-07-14:
Does anyone want to continue analyzing the problem?

AymericM78, 2020-07-15:
@roleoroleo  Me ! :)

roleoroleo, 2020-07-15:
Ok thanks.
Is this your current situation?
https://github.com/roleoroleo/yi-hack-Allwinner/issues/81#issuecomment-657665419

AymericM78, 2020-07-15:
yes, RTSP is KO and only work in low-res.
The issue appeared after 0.1.7 upgrade and even after unhack+downgrade to 0.1.6 the issue still there.
I have upgraded to 0.1.7.

For your information, high res snapshot is OK 

Let me know if you need to try something.

Thanks

roleoroleo, 2020-07-15:
I can't understand why it doesn't work with 0.1.6 after unhack.

Send me please a backup of your cam (only mtdblock3 that doesn't contain personal info).
And your configuration files (remove personal info).

roleoroleo, 2020-07-15:
And if you have time to spend, I would like to have the mtdblock3 after unhack + 0.1.6.

I tried yesterday with my cam and all is working properly.

murigny, 2020-07-16:
Hello.
I think i found the issue: Luminosity level
My 2nd cam is outdoor and have the issue with 0.1.7 but rtsp high only work when luminosity is less than 1000lx !

I made a test at 12h00 outdoor (no rtsp high): 
I'm covering the cam with a dark cover and wait a minute: rtsp high work
then, i remove cover: lot of artefacts and after a minute: no more picture.

@AymericM78 where is your cam ?
@roleoroleo  when you test, is your cam indoor? if yes, try to test outdoor.

roleoroleo, 2020-07-16:
Very well done, this explains many problems.
I will check it.
And yes, my cam is indoor.

AymericM78, 2020-07-16:
Sorry guys, I'm a little bit busy actually.
But I check your point @murigny, you're right my cam is outdoor and I confirm : luminosity is causing this issue!
Now it works!

So is there a way to fix this?
Because actually my cam is recording the wall and that not my goal! 🤣

roleoroleo, 2020-07-16:
I need some time to figure out if I can fix the problem.

roleoroleo, 2020-07-19:
I moved my cam outdoor and I have no problem.
Could you please run rRTSPServer in debug mode?
Kill rRTSPServer and kill wd_rtsp.sh.
Then run `rRTSPServer -r both -d`
I would like to know if the size of i-framws exceeds the buffer.

SirTerrific, 2020-07-20:
Hi, I had the problem with the upgrade from 1.6 to 1.7 . I tried several things (I had my cam continuously reboot but the last one which restore the high stream for me was that: 

- Unhack Procedure
- Install 1.6
- Disable RTSP server and restart
- Enable RTSP server and restart

After that still on 1.6, I have my High Resolution. It's pretty weird because I have 2 cam bought at the same time, installed and upgraded at the same time and only one have the but with the 1.7 . So I let it at 1.6 till a "new" version.

roleoroleo, 2020-07-20:
The continuous reboot is not related to rtsp issue.
Probably a configuration problem.

Muyayo22, 2020-07-20:
Good Morning,
I have the same problem too. I have two cameras outside, one of them gets plenty of light during the day, and I can't see it in high resolution while this is happening.
Thanks for the job!

murigny, 2020-07-20:
hello
howto output rRTSPServer debug into a log.txt ?
trying: "rRTSPServer -o both -d > log.txt" but log.txt is empty

roleoroleo, 2020-07-20:
The log is on stderr.
`rRTSPServer -o both -d 2>log.txt`

SirTerrific, 2020-07-20:
> The continuous reboot is not related to rtsp issue.
> Probably a configuration problem.
> 

I had this issue when i tried to downgrade to 1.6 the first time.

murigny, 2020-07-20:
> > The continuous reboot is not related to rtsp issue.
> > Probably a configuration problem.
> 
> I had this issue when i tried to downgrade to 1.6 the first time.

Yes, you have this issue if you downgrade without unhack before

roleoroleo, 2020-07-20:
See this issue: https://github.com/roleoroleo/yi-hack-Allwinner/issues/94

murigny, 2020-07-20:
Hello
You're right roleoroleo, framesize exceed buffer:

[log.txt](https://github.com/roleoroleo/yi-hack-Allwinner/files/4948658/log.txt)

roleoroleo, 2020-07-20:
Tonight I will send you a new version with a bigger buffer.

murigny, 2020-07-20:
Thanks, you're the best ;)

[19 more replies]

closed 2020-11-21 by roleoroleo