roleoroleo/yi-hack-Allwinner #84: Compile curl / busybox only for current camera

mfizz1, 2020-07-07:
I am a little new to compiling things, having looked at how to compile using the wiki here, it is only for compiling firmware and adding program to firmware which would take up space. I want to use programs on the sd card under yi-hack-v* folder.

I have the original yi home camera with yi-hack-v3 initially and now v4 on there there was:

busybox-1.28
curl

I can see that there has been an attempt made to compile on mstar also:

https://github.com/roleoroleo/yi-hack-MStar/issues/77

Is there a allwinner version?

roleoroleo, 2020-07-07:
You could use the scripts as a template and copy the output binary manually.

mfizz1, 2020-07-08:
Thanks for your response. I will attempt it and feedback.

mfizz1, 2020-07-08:
So I have tried to both extract the file and create symlinks myself manually, however I keep getting

"Operation not permitted"

```
ln -s /tmp/sd/yi-hack/lib/libcrypto.so.1.1 /tmp/sd/yi-hack/lib/libcrypto.so
ln -s /tmp/sd/yi-hack/lib/libcurl.so.1.1 /tmp/sd/yi-hack/lib/libssl.so
ln -s /tmp/sd/yi-hack/lib/libcurl.so.4.6.0 /tmp/sd/yi-hack/lib/libcurl.so
ln -s /tmp/sd/yi-hack/lib/libcurl.so.4.6.0 /tmp/sd/yi-hack/lib/libcurl.so.4
```
```
root@xiaoyi:/tmp/sd/yi-hack/lib# tar -xvzf curl-7.68.0.tar.gz
curl
libcrypto.so
tar: can't create symlink libcrypto.so to libcrypto.so.1.1: Operation not permitted
libcrypto.so.1.1
libcurl.so
tar: can't create symlink libcurl.so to libcurl.so.4.6.0: Operation not permitted
libcurl.so.4
tar: can't create symlink libcurl.so.4 to libcurl.so.4.6.0: Operation not permitted
libcurl.so.4.6.0
libssl.so
tar: can't create symlink libssl.so to libssl.so.1.1: Operation not permitted
libssl.so.1.1

```

Think it is just a permissions issue

mfizz1, 2020-07-09:
After some further looking I thought I might as well get everything else up and running, however it appears that the /tmp/sd/yi-hack folder on sd card isn't on the path by default? I thought it was looking at the "env.sh" and "system.sh" file. 

Reckon maybe the init.sh in home/app might be overriding the path and library. After manually setting the path, I am back at the original issue o being able to sym link

roleoroleo, 2020-07-09:
You can't create symlinks on FAT32 file systems. It's not supported.
As a workaround you could create 3 files for each library with different names and same content.

The PATH variable is changed by system.sh: all programs launched from system.sh inherit this variable.
But if you login with telnet or ssh you haven't it. You can add it temporarily (only for the current session) sourcing the file env.sh:
`. /home/yi-hack/script/env.sh`

mfizz1, 2020-07-10:
Okay, just did as you said, copied and renamed files. I suppose its not that big of an issue if on SD card anyway. I tested putting startup script with binaries on sd card and it seems to load the path and library just not in session. Not a big deal can just load the env.sh

closed 2020-11-21 by roleoroleo