roleoroleo/yi-hack-Allwinner #86: SSH File transfer not working

mfizz1, 2020-07-09:
In an attempt to reduce usage, I am disabling as many unnecessary services that I can.

So telnet, onvif, ftp. I want to continue using ssh and was hoping i could use it to transfer files as well. I can login using terminal and navigate around. Sometimes its just easier to transfer files and look at the tree structure. But whenever i try to connect something like filezilla / winscp I get an error.

WINSCP : exit code 127
filezilla: unexpected end of file from sftp-server

roleoroleo, 2020-07-09:
Create the following symlink:
`ln -s /home/yi-hack/bin/scp /home/base/scp`

Schm1tz1, 2020-07-10:
Same issue here with y20gam v0.1.7 even with the symlink. I am running with default users. SSH is working but no connection with frontends like Filezilla or Nautilus-SFTP in Gnome.

When using scp from the command line, -v gives me a hint:
...
debug1: Sending command: scp -v -f /home/yi-hack/bin/etc/hostname
ash: scp: not found
...

Seems to be a path issue I don't really understand right now. Even opening a new 'ash' has scp in path...

roleoroleo, 2020-07-10:
Probably the problem is that, when dropbear forks, doesn't inherit the environment.
And scp isn't in the standard PATH.
But the link suggested above works for me.

Schm1tz1, 2020-07-10:
Thank you very much! Just let me know if I can help with this issue in some way.

mfizz1, 2020-07-10:
I just tried it again and it doesn't work for me. Thanks for looking into it..

roleoroleo, 2020-07-10:
I tried again and it works without problems.
Check if the symbolic link is ok.

Schm1tz1, 2020-07-10:
Thanks for checking again. Does not seem to work for my case.
Symlink looks valid:
```
root@yi-d0e4:~# ln -s /home/yi-hack/bin/scp /home/base/scp

root@yi-d0e4:~# ls -l /home/base/scp 
lrwxrwxrwx    1 root     root            21 Jul 10 14:44 /home/base/scp -> /home/yi-hack/bin/scp

root@yi-d0e4:~# /home/base/scp 
usage: scp [-1246BCpqrv] [-c cipher] [-F ssh_config] [-i identity_file]
           [-l limit] [-P port] [-S program]
           [[user@]host1:]file1 [...] [[user@]host2:]file2
```
Same issue with scp:
```
$ scp -v root@yicam:/home/yi-hack/etc/hostname .
...
debug1: Sending command: scp -v -f /home/yi-hack/etc/hostname
ash: scp: not found
debug1: client_input_channel_req: channel 0 rtype exit-status reply 0
...
```

roleoroleo, 2020-07-10:
But it's to or from yi cam?

roleoroleo, 2020-07-10:
I don't know why.
If I run scp manually doesnt't work.
Winscp works.

roleoroleo, 2020-07-10:
Try this:
[dropbearmulti.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4904549/dropbearmulti.gz)

Schm1tz1, 2020-07-10:
> But it's to or from yi cam?

This is to yicam. I tried both directions, same error. 

> I don't know why.
> If I run scp manually doesnt't work.
> Winscp works.

Strange. The only thing that comes to my mind is possibly that the WinScp configuration is using FTP.

> Try this:
> [dropbearmulti.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4904549/dropbearmulti.gz)

I will try and let you know if it works for me, thanks !

Schm1tz1, 2020-07-10:
I installed the now dropbearmulti. 
Still ash does not find scp. I tried not with a copied version, not with symlink, not with mount --bind. Really very, very strange...

roleoroleo, 2020-07-10:
Try with new binary and symlink, together.

Schm1tz1, 2020-07-10:
Sorry, typo. That's what I did. Also tried with copies or bind instead of symlink. Somehow this path issue persists. I will use FTP or wget with a local webserver for now.

roleoroleo, 2020-07-10:
Tried exactly your command, it works:

```
root@omv:~# scp -v root@192.168.0.243:/home/yi-hack/etc/hostname .
Executing: program /usr/bin/ssh host 192.168.0.243, user root, command scp -v -f /home/yi-hack/etc/hostname
OpenSSH_7.9p1 Debian-10+deb10u2, OpenSSL 1.1.1d  10 Sep 2019
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to 192.168.0.243 [192.168.0.243] port 22.
debug1: Connection established.
debug1: identity file /root/.ssh/id_rsa type 0
debug1: identity file /root/.ssh/id_rsa-cert type -1
debug1: identity file /root/.ssh/id_dsa type -1
debug1: identity file /root/.ssh/id_dsa-cert type -1
debug1: identity file /root/.ssh/id_ecdsa type -1
debug1: identity file /root/.ssh/id_ecdsa-cert type -1
debug1: identity file /root/.ssh/id_ed25519 type -1
debug1: identity file /root/.ssh/id_ed25519-cert type -1
debug1: identity file /root/.ssh/id_xmss type -1
debug1: identity file /root/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_7.9p1 Debian-10+deb10u2
debug1: Remote protocol version 2.0, remote software version dropbear_2018.76
debug1: no match: dropbear_2018.76
debug1: Authenticating to 192.168.0.243:22 as 'root'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ecdsa-sha2-nistp256
debug1: kex: server->client cipher: aes128-ctr MAC: hmac-sha2-256 compression: none
debug1: kex: client->server cipher: aes128-ctr MAC: hmac-sha2-256 compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ecdsa-sha2-nistp256 SHA256:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
debug1: Host '192.168.0.243' is known and matches the ECDSA host key.
debug1: Found key in /root/.ssh/known_hosts:1
debug1: rekey after xxxxxxxxxxxxx blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey after xxxxxxxxxxxxxx blocks
debug1: Will attempt key: /root/.ssh/id_rsa RSA SHA256:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
debug1: Will attempt key: /root/.ssh/id_dsa
debug1: Will attempt key: /root/.ssh/id_ecdsa
debug1: Will attempt key: /root/.ssh/id_ed25519
debug1: Will attempt key: /root/.ssh/id_xmss
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentication succeeded (none).
Authenticated to 192.168.0.243 ([192.168.0.243]:22).
debug1: channel 0: new [client-session]
debug1: Entering interactive session.
debug1: pledge: network
debug1: Sending environment.
debug1: Sending env LANG = en_US.UTF-8
debug1: Sending command: scp -v -f /home/yi-hack/etc/hostname
Sending file modes: C0644 18 hostname
Sink: C0644 18 hostname
hostname                                                                                                                   100%   18     4.8KB/s   00:00
debug1: client_input_channel_req: channel 0 rtype exit-status reply 0
debug1: channel 0: free: client-session, nchannels 1
debug1: fd 0 clearing O_NONBLOCK
Transferred: sent 1856, received 1144 bytes, in 0.0 seconds
Bytes per second: sent 64766.6, received 39920.8
debug1: Exit status 0
```

Schm1tz1, 2020-07-10:
Thanks a lot for your help! After a reboot it works now at least in the command-line. The old service was still running in the background. 
Issue with FileZilla and Nautilus-SFTP in Gnome remains so I have to dig deeper.

roleoroleo, 2020-12-19:
Fixed in 0.2.2.

closed 2020-12-19 by roleoroleo