roleoroleo/yi-hack-Allwinner #87: Question: how to upload videos on Google Drive

marcotorti, 2020-07-09:
Hi all, I have been trying to use a script I found to automatically upload videos on gDrive. Unfortunately all my attempts to trigger the script via telnet failed.
Is there an alternative way to do so? Any recommendations?

mfizz1, 2020-07-09:
what error are you getting?

marcotorti, 2020-07-09:
I cannot run the script from porunov:
https://github.com/porunov/xiaomi_gdrive

I know it is for a different Yi Cam version, I was just hoping it could be a starting point.

I saw there has been quite a lot of work on this in the MSTAR version of the camera (https://github.com/sorokin-e/xiaomi_gdrive), but not sure it is compatible with the Allwinner version..

I apologise for my limited knowledge, I just started with Yi cameras, but I'm reading everything I can find.

mfizz1, 2020-07-09:
You need to be a little more specific to what the error is.. what message are you getting when you run the script

roleoroleo, 2020-07-09:
> I saw there has been quite a lot of work on this in the MSTAR version of the camera (https://github.com/sorokin-e/xiaomi_gdrive), but not sure it is compatible with the Allwinner version..

This is a good start point. Probably there are some busybox commands to change.
And you need this archive:
[gdrive_utils.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4899398/gdrive_utils.tar.gz)

marcotorti, 2020-07-10:
Thank you both, I'll give it a try and keep you informed

Mick2011, 2020-07-29:
Did anyone get this working?

Mick2011, 2020-08-08:
:(

mfizz1, 2020-08-08:
Have you tried it yet? It appears there is no reason why it shouldn't work.. there is a compiled curl and busy box and the script from mstar version has been consolidated to work with the limited number of  binaries available..

Mick2011, 2020-08-08:
I didn't, I'll give it a try tomorrow if I can figure it out and report back.

closed 2020-10-06 by roleoroleo