roleoroleo/yi-hack-Allwinner #88: Feature-Request: Please add functionality for uploading firmware .tgz

Schm1tz1, 2020-07-10:
A nice-to-have feature would be the possibility of uploading a specific version or your own compiled firmware (e.g. for testing) using the web interface:

Idea: refactor the fw_upgrade.sh
- allow configurable URL as a firmware source (so you can have your local FW repo) wit default=(latest release on GitHub)
- allow upload of .tgz via web interface

roleoroleo, 2020-07-14:
It would be a nice feature.
But a lot of work...
I will try to find the time.

Schm1tz1, 2020-07-14:
I have at started to extract the pull-mechanism of the FW into a script that simply takes the URL as a argument in a [separate branch](https://github.com/Schm1tz1/yi-hack-Allwinner/tree/firmware_upload). That way you can manually update the FW from a custom URL via SSH in a first step and have some "unified installation". You can simply host the FW in you local network with e.g. `python -m SimpleHTTPServer` and then install via `$YI_HACK_PREFIX/script/pull_and_install_firmware.sh <FIRMWARE_URL>`. That seems to work as a first step.

Schm1tz1, 2020-07-14:
Next step would be to allow installation from a local file so you can perform the same installation from an uploaded file. Here I have tried uploading the FW based on the "Configuration Load" routine but ran into performance issues. The cat | awk - Loop that is done is not really optimal as for every single line the whole file is processed just to count the bytes between the beginning and end marker. For the configuration that is fine but for 3 megabytes of firmware it takes ages... 
So far I haven't found a better solution just using cat, sed, awk and haveing the whole file caches in memory is not an option. Normally you have only a few megabytes of free RAM...
Sed is problematic for binary files because of the CR and LF characters. In the best case I ended up with a .tar.gz that has 0D 0A in the end and thus tar will extract but end with an error.
One could either think about doing the extraction directly in JavaScript or to do something based on wc/head/tail (which is not yet included and needs to be compiled).

roleoroleo, 2020-07-14:
Probably we have to write a C program.
I think we could find something useful in internet.

closed 2024-05-13 by github-actions[bot]