roleoroleo/yi-hack-Allwinner #89: No MQTT start message

Darios77, 2020-07-10:
Hi,
I flashed today Yi 1080p camera with latest firmware 0.1.7
All works nice, just MQTT sends only stop message. Start is not sent.
I configured it to work with HA, but also monitored with MQTT Explorer. And only STOP message I can see.
Looks like I'm the first one with this problem, so question is - what I did wrong? :)
Regards.

roleoroleo, 2020-07-10:
See this:
https://github.com/roleoroleo/yi-hack-MStar/issues/189

Darios77, 2020-07-14:
Hi,
took me some time to start testing. 
From what I understand, what helped was to reset the camera, reflash to older firmware and update.
I did that.
Several times in fact. To different versions. 0.1.4 to 0.1.6
Tried even to updated in sequence from 0.1.4 to 0.1.7
No luck.
What's more - only version 0.1.7 works normally.
On olders fw camera is accessible for few seconds and after is not connected to network (maybe even restarts, not sure. Sometimes are clicks inside, sometimes no. And I waited quite long to be sure it finished updating. When I refreshed in time web page, I've seen the fw version I selected, so I guess was flashed successfully). 
After several seconds comes back for few pings and disconnects again.
Still - on 0.1.7 is ok.
Just no start message in MQTT :)

roleoroleo, 2020-07-15:
Probably the problem in the previous version is related to this issue: https://github.com/roleoroleo/yi-hack-Allwinner/issues/94
About the MQTT start message, try to change some settings in the yi app.
I suspect there is a problem there.

Bodengriller, 2020-07-29:
I can also confirm this Error.
I have 2 devices with the same effect (Yi Home 1080p 9FUS).
Both devices do not send a start message, only stop and motion_files.
I have already reset both devices (button on the back), added them completely new in the app and set them up again, I have tried different yi-hack-allwinner versions (0.1.4 to 0.1.8)
I have updated with the SD card as well as tried the online update - no change. :(

There seem to be a few people who have this problem after all - for many of them, it will magically work for them at some point.

OffTopic:
Whatever I notice, and I'm still not sure where it comes from:
When I scroll through the past in the app and want to watch with recordings in which a movement has been registered (the timeline is then orange instead of brown), there are always huge jumps...you never land where you want to land (usually much further in the past) and there is no way to "fast forward" further - it always jumps back.
I guess this has little to do with this firmware hack?
(Same effect on both devices)

Bodengriller, 2020-07-30:
After a restart (without power) it seems that no more MQTT messages are sent.

Now I grabbed one of the two cameras to "play":
```
1. unhack
2. reset (button on back)
3. completely new in the app
4. yi-hack-Allwinner 0.1.4 installed (with SD card)
5. MQTT established
5a. MQTT does not work!
6. upgrade to 0.1.8 via webfrontend
6a. MQTT does not work
7. reset (button on back)
8. completely new in the app
9. MQTT works! (start and stop messages !)
9a. cheers :)

10. reboot (camera carried to another place)
11. no more mqtt messages are sent
12. reset + reconfig
13. no mqtt messages
```
:(

I have read a lot here in the Issues and could provide the following logs:

root@Stellplatz:\~# **killall mqttv4**
root@Stellplatz: \~# **/home/yi-hack/bin/mqttv4**
```
Error loading shared library libmosquitto.so.1: No such file or directory (needed by /home/yi-hack/bin/mqttv4)
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_connack_string: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_publish: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_publish_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_connect_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_log_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_username_pw_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_lib_cleanup: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_new: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_connect: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_strerror: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_disconnect_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_loop: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_lib_init: symbol not found
```

root@Stellplatz:\~# **killall mqttv4**
root@Stellplatz:\~# **. /home/yi-hack/script/env.sh**
root@Stellplatz:\~# **/home/yi-hack/bin/mqttv4**
```
Starting mqttv4 v0.0.3
key: MQTT_RETAIN | value: 1
Unrecognized config.
Trying to connect... Client testcam sending CONNECT
Client testcam received CONNACK (0)
. connected!
Client testcam sending PINGREQ
Client testcam received PINGRESP
```

root@Stellplatz:\~# **killall mqttv4**
root@Stellplatz:\~# **LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/yi-hack/lib**
root@Stellplatz:\~# **/home/yi-hack/bin/mqttv4**
```
Starting mqttv4 v0.0.3
key: MQTT_RETAIN | value: 1
Unrecognized config.
Trying to connect... Client testcam sending CONNECT
Client testcam received CONNACK (0)
. connected!
```

root@Stellplatz:**/home/yi-hack/etc# vi mqttv4.conf**
```
###############################################################################
#                       Configuration file for mqttv4                         #
###############################################################################

# -----------------------------------------------------------------------------
# Basic MQTT settings
# -----------------------------------------------------------------------------

MQTT_IP=192.168.178.11
MQTT_PORT=1883

MQTT_CLIENT_ID=testcam

MQTT_USER=mosquitto_user
MQTT_PASSWORD=removed

MQTT_PREFIX=yicamtest

# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_MOTION=motion_detection
TOPIC_MOTION_FILES=motion_files
TOPIC_BABY_CRYING=baby_crying

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop
BABY_CRYING_MSG=crying

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN=1

MQTT_RETAIN_MOTION=1
MQTT_RETAIN_MOTION_FILES=1
MQTT_RETAIN_BABY_CRYING=1
```

**Camera-Settings:**
```
Switch on/off the camera: On
Save video when a motion is detected: On
Detection sensitivity: Medium
Baby crying detect: Off
Status led: Off
IR led: On
Rotate: Off
```

The camera lies beside me and I also move occasionall :D. 
So movements were detected - in the app these were also detected correctly and recorded on the SD card.
But apparently this was not transmitted to mqtt

Any idea what else I could do?

_PS (OffTopic):
The jumps when playing in the app are probably related to the timezone and the NTPD.
Something messes up the time formats of the recordings during reboot of cam, so that e.g. some recordings in the timeline are marked as "recorded" (orange instead of brown) in the future.
That's when the app gets confused during playing.
If I leave the timezone empty and deactivate the NTPD, everything works as it should...but here I will continue to research further, once MQTT is running properly and make a seperate issue._

roleoroleo, 2020-07-30:
> There seem to be a few people who have this problem after all - for many of them, it will magically work for them at some point.

I think it depends on some yi app configuration.
But I don't know how to help you.

> OffTopic:
> Whatever I notice, and I'm still not sure where it comes from:
> When I scroll through the past in the app and want to watch with recordings in which a movement has been registered (the [...]

Did you change the timezone?

roleoroleo, 2020-07-30:
Remove MQTT_RETAIN in your config file and try to capture the log again.

But probably we will not see detected motions.

Bodengriller, 2020-07-30:
> But I don't know how to help you.
Hmpf that's too bad.
Based on the logs, do you have any other ideas where you could have a look?

> Did you change the timezone?
Yes, I changed it to "CET-1CEST,M3.5.0,M10.5.0/3" (Europe/Berlin) ...
I suspect that the phenomenon is more related to this than to the NTPD - I simply deactivated the NTPD because I don't need it for now (no private mode)

Bodengriller, 2020-07-30:
> But probably we will not see detected motions.

Unfortunately, you're right.

root@Stellplatz:~# killall mqttv4
root@Stellplatz:~# LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/yi-hack/lib
root@Stellplatz:~# /home/yi-hack/bin/mqttv4
```
Starting mqttv4 v0.0.3
Trying to connect... Client testcam sending CONNECT
Client testcam received CONNACK (0)
. connected!
```

With which app settings does it work (for you?) in any case?
Where could you start to find the error?

roleoroleo, 2020-07-30:
Motion start is missing.
Same behavior as other users.

bigjohnson, 2020-08-04:
We also have the same problem, with 5 camera with 0.1.8 version, we try with factory reset, with no luck.
Thanks.

roleoroleo, 2020-08-05:
Try to sniff ipc messages using this tool:
[ipc_sniff.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5026591/ipc_sniff.tar.gz)

- kill mqttv4 and ipc_multiplexer.
- source environment: `. /home/yi-hack/script/env.sh`
- run ipc_sniff
- post the log

Real-Konai, 2020-08-13:
After playing a little bit with mqtt and the Yi Home App, it seems that this is related to the person detection in Yi Home.
If person detection was enabled, no motion_start is sent in MQTT, even if motion detection is disabled. With person detection disabled the motion_start message is sent regularly.

Hope, this helps a little bit.

roleoroleo, 2020-08-13:
Thanks for your clarification, I've been trying to solve this problem for months.

Darios77, 2020-08-13:
I did not suspect this option to be the problem. In fact human recognition was something interesting for me, so I didn't think to turn it off.
And it helps. After turning it off, in MQTT both messages are present.
Thanks.

74ls04, 2020-09-04:
Any clues as to why the human detection setting would stop the start messages? I would be willing to help troubleshoot if I can be pointed in the right direction to start.

The human detection is probably one of the more useful features and the short video it sends is quite helpful to quickly evaluate the alert!

zenoran, 2020-12-12:
Anyone figure this out? My cameras don't have human detection setting but for some reason they all stopped sending motion start events.

Matssa56, 2020-12-23:
FYI, just tested this out and there is at least something clear : it isn't linked to the disable cloud feature. Indeed with or without it the issue is still there. Deactivate human AI detection and the motion will work. If you activate it, it won't work anymore.

Apparently same issue using YI app.

Gonna return the 2 that I bought since it was purchased for human detection (in order to avoid detecting my cats...). Gonna follow this in case a new YI firmware fixes this (let's hope!)

colethegamer123, 2021-01-30:
Hello,

I disabled Person detection and Motion Detection in Yi Home. I installed the latest firmware to my Cameras. I setuped MQTT.

Motion Detectin: Working
AI Human Detection: Not working.

Any Idea how can I solve this issue?

```
binary_sensor:
  - platform: mqtt
    name: "Motion detection - Kitchen"
    state_topic: "home-assistant/kitchen/motion_detection"
    payload_on: "ON"
    qos: 1
    off_delay: 10
  - platform: mqtt
    name: "AI Motion detection - Kitchen"
    state_topic: "home-assistant/kitchen/human"
    payload_on: "ON"
    qos: 1
    off_delay: 10
```

[image]

Thank you very much!!!

colethegamer123, 2021-01-30:
I also bough this Camera to avoid Cats.

roleoroleo, 2021-01-30:
Sorry but I think this feature is not working.
Both with the original firmware and with the hack.

colethegamer123, 2021-01-30:
> Sorry but I think this feature is not working.
> Both with the original firmware and with the hack.

Thank you very much for your fast answer and of course for your hard work. I'll try out image processing for that. Have a nice day.

Antoni-Czaplicki, 2021-02-21:
Is there any solution to this problem?

colethegamer123, 2021-02-23:
> Is there any solution to this problem?

As I know the answer is : No

dallas6672, 2022-05-01:
I just started getting this issue with 0.3.0. I was previously on either 0.2.7 or 0.2.8. The only thing that seems to fix it, is to re-enable the Cloud. So, as long as Cloud is enabled it works. As soon as I disable the Cloud again, it stops sending the "motion_start". 

I've tried disabling and re-enabling Motion from the app. Then turning the Cloud back off, but that does not fix the issue either.

dallas6672, 2022-05-02:
Uninstalling the mod and downgrading to each version all the way back to 0.2.6 finally worked.

roleoroleo, 2022-05-05:
Is the cam filtered with a firewall?

roleoroleo, 2022-05-07:
Are you using human detection?

dallas6672, 2022-05-07:
Never used the human detection. I did try to turn that on when I was having the issue to see if that would work, but it didn't either. 

Sorry, forgot to come back here last night and give an update after I upgraded to 0.3.0 again. So far it seems to be working fine.

dallas6672, 2022-05-10:
It is running properly now

dallas6672, 2022-09-27:
> I just started getting this issue with 0.3.0. I was previously on either 0.2.7 or 0.2.8. The only thing that seems to fix it, is to re-enable the Cloud. So, as long as Cloud is enabled it works. As soon as I disable the Cloud again, it stops sending the "motion_start".
> 
> I've tried disabling and re-enabling Motion from the app. Then turning the Cloud back off, but that does not fix the issue either.
> 
> edit...Just noticed that this is also causing it to not record motion events. If I enable Cloud and Block Internet traffic, then "motion_start" works, but still no recording and RTSP fails (#234) and did not receive a "motion_stop"
> 
> Base Version | 8.2.0.0A_202007211802 Model Suffix | y20ga

This issue seems to have returned. Can't seem to get it working again like I did last time by removing and reflashing the mod. Went as far back as 0.2.4 this time. 

As long as the Cloud is NOT disabled it works. I can even block Internet access while the Cloud is enabled and it works. Been working ever since the May 5th. Flashed 0.3.1 several days ago it was working fine until today around Noon it had a motion detection and that was the last one.

roleoroleo, 2022-09-30:
Are you able to start the process manually and check the output?
Login with ssh and run these commands:
```
killall mqttv4
. /tmp/sd/yi-hack/script/env.sh
mqttv4
```

dallas6672, 2022-09-30:
For some reason it just randomly started working again. But when the issue was happening, I don't think it was mqtt since the Online and Offline messages were still being sent. Also, the recording on motion detection also stopped working. It's almost like it just stopped detecting motion altogether.

roleoroleo, 2022-10-01:
So you don't get events in the app either.
Right?

dallas6672, 2022-10-01:
> So you don't get events in the app either. Right?

Correct

closed 2024-05-13 by github-actions[bot]