roleoroleo/yi-hack-Allwinner #93: Update brick the camera

MrPatate87, 2020-07-14:
Just see today you made some update to this great hack.
Loving this work and whant to find the new features on this hack I decide to made the update via the manadgement tool of the web interface. 
My previous FW was 0.1.6 version, i clic to the update link of the interface telling me downloading new verqion and rebooting the camera.
But the damera never reboot, ear a clic, and the led always on on yellow, but nothing more.
When I trying to reconnect to the web interface, no result, th rtsp I have se on this cam was not connected.
What can I do to made this cam came back to life?

MrPatate87, 2020-07-14:
Really sorry, the update was made after a long time so long I really think the camera was brick (More than 15 minutes) but seems run well

closed 2020-07-14 by MrPatate87

roleoroleo, 2020-07-15:
This is a know issue.
If the SD card is full (eg a lot of mp4 files) the upgrade requires more time.