roleoroleo/yi-hack-Allwinner #95: Enhancement: multi view

Merlynda, 2020-07-15:
The Yi app offers multi view. But to view more than one camera you need to subscribe to cloud service. Anyway, to get multi view without subscription? -had to ask.

murigny, 2020-07-15:
If your hack your cam, you can access rtsp stream.
And then, you can use apps like ispy64, contacam, motion, etc ... that offer multiview.

Merlynda, 2020-07-15:
Yes I tried IP Cam Viewer. It was OK. If that’s the only way for multi view the IP cam viewer will work. But having multi view in Yi app Without paying would make me really happy

roleoroleo, 2020-07-15:
Sorry but to have multi view in Yi app you have to pay.
It's a cloud feature, indipendent from the firmware.

But, as @murigny wrote, you can use a lot of program to do this.
With a server you can install motion or zoneminder, for example. Or a domotic solution like home assistant.
If you want to use your smarphone: IP cam viewer, tinycam, onvifer, etc...

Merlynda, 2020-07-15:
Darn!

closed 2020-07-15 by Merlynda