roleoroleo/yi-hack-Allwinner #96: Blue flashing light

mattpreston11, 2020-07-15:
Hey, First off thank you for your work.

I have a model 9FUSY31X9I9EK4200319 with firmware 8.2.0.0A_201912270941.

Copied the files to multiple FAT32 SD cards (2GB + 32GB). camera just has a blue flashing light forever. Never connects to my local network. Tried firmwares Y20gam_0.1.7, Y20gam_0.1.6, Y20gam_0.1.3

Removing SD card and rebooting, Camera connects fine and is available in YI app. However still no connection to web page or telnet.

Any ideas?

roleoroleo, 2020-07-15:
Did you respect the directory structure?
Factory and newhome?
With the correct upcase/downcase?

mattpreston11, 2020-07-15:
Hey, I have copied the files directly onto the SD. 

[image]

roleoroleo, 2020-07-15:
I don't know.
Reformat the sd and try again.

closed 2020-11-21 by roleoroleo