roleoroleo/yi-hack-Allwinner #97: 9FUSY312E5 not flashing

andker87, 2020-07-17:
Hi there, just got a Yi 1080p Home Camera 9FUSY312E5, but no way to flash it...tried all the ways, even the backup procedure, but no results (no backup too).
I tried with three different microsd, but all failed.

Can someone help me? Or maybe can I help in any way to add this model to the compatibility list? thanks

roleoroleo, 2020-07-18:
Check the fw version with yi app.

andker87, 2020-07-18:
> Check the fw version with yi app.

firmware 8.2.0.0A_201912270941

roleoroleo, 2020-07-18:
Probably the same issue described here:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/59

mattpreston11, 2020-07-22:
> > Check the fw version with yi app.
> 
> firmware 8.2.0.0A_201912270941

I have the same version, Do you get a constant blue flashing light with the SD card?
I've ordered a new camera and ended up with another 9FUS31 (with different characters after). Got exactly the same issue as the first one.,

[https://github.com/roleoroleo/yi-hack-Allwinner/issues/96](url)

andker87, 2020-07-22:
> 
> I have the same version, Do you get a constant blue flashing light with the SD card?
> I've ordered a new camera and ended up with another 9FUS31 (with different characters after). Got exactly the same issue as the first one.,
> 

Yes, I get a constant blue flashing light, the yellow light is just for few seconds at startup, then the solid blue led.

I also have another cam (9FUS310 if I remember well), and I'm stuck with the same problem.

roleoroleo, 2020-07-22:
Are you able to get a dump of the flash?

mattpreston11, 2020-07-22:
> Yes, I get a constant blue flashing light, the yellow light is just for few seconds at startup, then the solid blue led.
>
> I also have another cam (9FUS310 if I remember well), and I'm stuck with the [...]

Hopefully we get a fix in future firmware. Or someone who has had this issue can explain how they got past it.

andker87, 2020-07-22:
> Are you able to get a dump of the flash?

no, unfortunately the backup procedure does not work

andker87, 2020-07-22:
> Hopefully we get a fix in future firmware. Or someone who has had this issue can explain how they got past it.

the only thing that comes to mind (but that I can't try right now) is to format the microsd directly from the YI CAM app, then insert it in the computer and copy the files and try it

mattpreston11, 2020-07-22:
> Are you able to get a dump of the flash?

Hey, Tried the firmware dump documented here

 [https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card)](url)

Exactly the same behaviour, 8 seconds of yellow light, Then blue flashing light forever.

Tried formatting multiple SD cards in the app and pasting files, Always blue flashing light.
When removing the sd card, or using an sd card without the hack files, Camera boots up and connects to wifi fine.

roleoroleo, 2020-07-22:
And what about a serial dump?

mattpreston11, 2020-07-22:
> And what about a serial dump?

I've ordered a soldering iron, I'll let you know 👍

andker87, 2020-07-22:
> And what about a serial dump?

@roleoroleo got serial connection, do you need just the boot log or the home partition?

roleoroleo, 2020-07-22:
Both  if you can.

andker87, 2020-07-22:
got the log, the home and the whole flash. how can I share with you?

bonuzzz, 2020-07-22:
blue flashing means camera is waiting for wifi credentials via camera (yi app)

bonuzzz, 2020-07-22:
@andker87 what microsd do you use?

andker87, 2020-07-22:
@bonuzzz I tried 5 different microsd, 2x Sandisk Ultra, Netac, Kingston and SP...
how can I check the folder existance?

bonuzzz, 2020-07-22:
if you have solid blue light, your camera works properly.
on your backup open file mtdblock3.bin and find word "yi-hack". if you find any mentions, so hack has been flashed

andker87, 2020-07-23:
The camera is working, that is not a problem. The problem is that I can't apply the hack, so I conected via serial to give more infos
I followed [this](https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(serial-cable)) procedure and I got a home.bin backup file, but there is no any mtdblock3.bin.
if I try to open the home.bin file with any text editor (notepad++) what I get is unreadable, so how I can open that file?

bonuzzz, 2020-07-23:
Use this method and open with notepad++ https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card)

andker87, 2020-07-23:
> Use this method and open with notepad++ https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card)

That method does not work, that is the reason why I connected the cam via serial.

bonuzzz, 2020-07-23:
Then I very doubt the hack flashed also.
I have the same two cameras 9fusy31. they flashed without problem but I use old trancend microsd class4 4 gb
I have one idea
https://drive.google.com/file/d/1U3DmzhHljD8sRVKJlBl-Vv5g7eJCTt07/view?usp=sharing
paste Factory folder to microsd. If hack method works you should receive test.xt file on microsd
If the method will not work, try to catch log with microsd inserted. In your log above microsd had been detected. And if you have old low capacity microsd, try also that one.

mattpreston11, 2020-07-23:
> Then I very doubt the hack flashed also.
> I have the same two cameras 9fusy31. they flashed without problem but I use old trancend microsd class4 4 gb
> I have one idea
> https://drive.google.com/file/d/1U3DmzhHljD8sRVKJlBl-Vv5g7eJCTt07/view?usp=sharing
> paste Factory folder to microsd. If hack method works you should receive test.xt file on microsd
> If the method will not work, try to catch log with microsd inserted. In your log above microsd had been detected. And if you have old low capacity microsd, try also that one.
> 
> upd: I changed file permissions in gdrive.

Hey,

Tried this on mine and it worked! First time any of the files have done anything other than give me a blue flashing light

[image]

andker87, 2020-07-23:
@bonuzzz thank you, I'll try this eveninng and give you a feedback about that.

bonuzzz, 2020-07-23:
@mattpreston11 solid blue light or blink?
when you flash the hack, folder Factory renames to Factory.done?

roleoroleo, 2020-07-23:
@bonuzzz 
What the difference between normal hack procedure this one?

mattpreston11, 2020-07-23:
> @mattpreston11 solid blue light or blink?
> when you flash the hack, folder Factory renames to Factory.done?

Hey, When I flash your testhack I get solid blue light, Camera is connected to wifi, And the folder structure looks like this:
[image]

However nothing with the firmware has changed which is expected? This is just a simple test I guess.

When I try to flash the Y20gam_0.1.7 with the following folder structure, I get the same 8 seconds of yellow light followed by indefinite flashing blue light and the camera never connects to wifi. After waiting 5 minutes and checking the files, they haven't changed.

[image]

mattpreston11, 2020-07-23:
> @bonuzzz
> What the difference between normal hack procedure this one?

Hey, Seems this is just a test to see if the camera is reading the card correctly and making two changes to verify

`echo 1 > /tmp/sd/test.txt

if [ -e /tmp/sd/Factory ]; then
    mv /tmp/sd/Factory /tmp/sd/Factory.done
fi

reboot
`

bonuzzz, 2020-07-23:
@roleoroleo this script just tests working status of "exploit" and compatibility of microsd. I have thoughts that high capacity of high speed microsd could be incompatible with camera.

sulivan22, 2020-07-23:
Hello all.

     We dont have resolution for the firmware 8.2.0.0A_201912270941 yet? i try all procedures that i found but dont work with my camera :( 

Regards

bonuzzz, 2020-07-23:
@mattpreston11 
https://drive.google.com/file/d/1VZHYrWoGaREVCMHMxipdEZV_qpr1cXj6/view?usp=sharing
this is hack fw 0.1.7gam with debug. try to flash and check files on microsd after that.

sulivan22, 2020-07-23:
Hello all

     Finally works, with the process of yi-hack-Allwinner, you have to be patience the process take time, and them you can in to the camera via we (http://192.168.1.XX:8080/)

     i try all the rest of process to flash the yi 1080p Home with FW firmware 8.2.0.0A_201912270941 but just work the yi-hack-Allwinner process.

Thanks

mattpreston11, 2020-07-23:
> @mattpreston11
> https://drive.google.com/file/d/1VZHYrWoGaREVCMHMxipdEZV_qpr1cXj6/view?usp=sharing
> this is hack fw 0.1.7gam with debug. try to flash and check files on microsd after that.

Hey @bonuzzz copied these files and the hack has worked! webpage, telnet etc. 
What changes have you made? Do you still want my debug files? I guess they won't show much.

bonuzzz, 2020-07-23:
I just added echo strings like echo 1 > /tmp/sd/1.txt to config.sh to check where exactly script stucks.
No, I don't want your files, because if everything os working now, flash script executed successfully.

andker87, 2020-07-23:
> https://drive.google.com/file/d/1U3DmzhHljD8sRVKJlBl-Vv5g7eJCTt07/view?usp=sharing
> paste Factory folder to microsd. If hack method works you should receive test.xt file on microsd

> I have thoughts that high capacity of high speed microsd could be incompatible with camera.

SOLVED: it was the microsd
I used 5 different high-speed microsd, than I tried an old 128mb one and it worked, I received the test.txt, so I tried with the hack folder and it worked.

roleoroleo, 2020-07-23:
Very strange.
I normally use a 8GB or a 2GB SD.
Never had any problems.

@andker87 solved with a different SD.
@mattpreston11 solved using the hack proposed by @bonuzzz (only debug echo added).
@sulivan22 solved waiting more time.

Other issues?

andker87, 2020-07-23:
This evening I can give you more details about each sd if you need.

(little suggestion: in the "dump via serial interface" wiki, specify that the baudrate proposed by putty has to be changed...for me the right one is 115200)

pbanj, 2020-07-27:
I use these https://smile.amazon.com/dp/B073K14CVB/ in each of mine and haven't had an issue flashing them. Have 4 cameras and 4 of those SD cards.

MaxWinterstein, 2020-08-09:
Just to drop my 2 cents here too:  
Bought one on Amazon DE 2 weeks ago, `9FUSY31SQ3P...` and had the same problems. After flashing to `0.1.3` i was able to flash towards `0.1.8`. No idea why.

Thanks for that awesome firmware!

closed 2020-10-06 by roleoroleo

trash-anger, 2021-02-05:
> @mattpreston11
> https://drive.google.com/file/d/1VZHYrWoGaREVCMHMxipdEZV_qpr1cXj6/view?usp=sharing
> this is hack fw 0.1.7gam with debug. try to flash and check files on microsd after that.

Thanks !