roleoroleo/yi-hack-Allwinner #98: How ot be sure to get the good camera ? (serious issue ! :p)

skrew, 2020-07-18:
Hi,

Sorry, i know it's not a dev question, but I've already bought 2 Yi dome 1080p and got two 2016 version (i wanted 2019/2020 version for MStar firmware)

As this repository is active, i want to test with supported Yi home 1080p with this firmware.

It's safe to buy a Yi home from Amazon(.de) ?

Thanks !

roleoroleo, 2020-07-18:
I think it's not possible.

jjansen85, 2020-07-18:
I bought it from amazon.nl. Got a 9FUSY310OZ
It is running perfectly with this firmware.

Catfriend1, 2020-07-19:
Amazon germany also delivered a compatible model to me of the 1080p home a month ago.

skrew, 2020-07-19:
@Catfriend1 Can you confirm it's this one ? 

https://www.amazon.de/YI-%C3%9Cberwachungskamera-Nachtsicht-Bewegungsmelder-Smartphone/dp/B01MR82NFZ/ref=sr_1_1_sspa?dchild=1&keywords=yi+home&qid=1595161756&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExN1lETklPNTk5SEZVJmVuY3J5cHRlZElkPUEwNTEyNjk4TUpEMThUREpHVDVWJmVuY3J5cHRlZEFkSWQ9QTAwMTk4OTkyQUFRNkdGVkxYMllQJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

Thanks !

Catfriend1, 2020-07-19:
We've bought the "family pack" , Link https://www.amazon.de/gp/aw/d/B06XHVNNNV/ref=psdcmw_119915031_t1_B01MR82NFZ

Schm1tz1, 2020-07-21:
@skrew I have bought 2 of these family packs. Both contained 9FUSY3* cameras that work perfectly with the Allwinner-Firmware.

Catfriend1, 2020-07-21:
Interesting, mine is a BFUSY model

Schm1tz1, 2020-07-21:
Checked again, same link, same article, all are 9FUSY. I bought one 2 weeks ago and one yesterday.

Catfriend1, 2020-07-21:
Yes, I believe you :). Wonder why there is such a model variety out there...

skrew, 2020-07-21:
Thanks all, i've ordered on amazon.de, i will tell you

skrew, 2020-07-22:
Received (9FUSY31)
First attempt, i got a blink yellow, and after a reset, a solid yellow.
Second try was the good one !

Image look "too much detailed", like after applied sharpening filter in photoshop too many time (i don't know if it come from the hack or if it's the normal quality) but it works and now i have my timelapse camera, finally ! :)

Thanks !

closed 2020-07-22 by skrew

roleoroleo, 2020-07-22:
It depends on the camera, not on the hack.