roleoroleo/yi-hack-Allwinner #99: Did something change with SSH?

koichirose, 2020-07-21:
I can't login via SSH anymore (root/empty pw).
I just updated to 0.1.7 from 0.1.6, hadn't turned on the camera for a while.

I see that you can now change your SSH password (how?).
I don't think I touched anything else.

Thanks!

roleoroleo, 2020-07-21:
Try to clean the cache of the browser and then reset to yi-hack default.

koichirose, 2020-07-22:
It worked.
I got it, the username is root, the password is what you choose for the web interface password.
Maybe this should be added in the web interface:
` Password to access RTSP, ONVIF and HTTP (default empty). `
should be
` Password to access RTSP, ONVIF, HTTP and SSH (default empty). `

closed 2020-07-22 by koichirose

roleoroleo, 2020-07-22:
This behavior will be changed in the next version.
I added a new text field.