roleoroleo/yi-hack-Allwinner #101: FTP upload - sub directory option

divadiow, 2020-07-29:
Hi
Would it be easy to add an option to choose a subfolder for ftp upload as at the moment it's only to the root.

Cheers

Catfriend1, 2020-08-02:
Hi,
I don't know if ftpput can create directories to be reliable there...(?)
Maybe a prefix would be a better idea, like the camera name so you can filter the uploads of multiple cameras.

roleoroleo, 2020-08-02:
ftpput supports remote name option.
Where you can use a path with subfolders
Like:

```
ftppush -u user -p password destdir/remotefilename.mp4 localfilename.mp4
```

divadiow, 2020-08-02:
Ah ok cool. It was more about uploading to ftp that doesn't let you save to the root, rather than organising files. Lots of routers with FTP facility for sharing files off a USB hard drive appear to put the writeable space under a subfolder, a folder of the drive name or something like "Drive1".

Catfriend1, 2020-08-16:
I think I'll for a new PR soon to add subdir and prefix to the setrings UI.

divadiow, 2020-08-18:
oh great. thanks

stsirakidis, 2020-08-18:
Btw, would it possible to add FTPS? ...in order to connect to remote FTP server over a secured connection?

roleoroleo, 2020-08-18:
No, what about sftp?

stsirakidis, 2020-08-19:
> No, what about sftp/scp?

Yes, it would be great, if you could add them!

roleoroleo, 2020-08-19:
dropbear is already compiled in the hack.
But I didn't test it as a client.
Di you want to try?

stsirakidis, 2020-08-19:
I tried dbclient as SSH client.  It got stucked by trying to connect to a remote SSH server - following warning was also displayed:
./dbclient: Warning: failed creating /root/.ssh: Read-only file system

Regarding scp, it seems that it is "mislinked" to **/usr/yi-hack/bin/dbclient**:
root@XXXXXX:/home/yi-hack# bin/scp model_suffix XXXXXXXXXXX
/usr/yi-hack/bin/dbclient: No such file or directory

Anyway...sorry...I didn't want to change the purpose of this issue. Regarding this issue from divadiow...I would also like to have  a subfolder option in the "FTP Upload" section within the  "Motion Events".

I would suggest that I will open a new issue, regarding SCP/SFTP....as soon as this issue is implemented in a new major release... Just keeping things easy ;-)

roleoroleo, 2020-08-19:
@stsirakidis Regarding dbclient there was an error in localoptions.h
I solved it and now it works.
If you want to try it, overwrite dropbearmulti with this:
[dropbearmulti.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5096896/dropbearmulti.tar.gz)

PR #130 (2020-09-02): ftppush: Allow specifying target directory (fixes #101)
- ftppush: Allow specifying target directory (fixes #101)
- Create target dir if it does not exist

Verified working on my device.
commit 318db691 (2020-09-02): Merge pull request #130 from Catfriend1/ftpdir

ftppush: Allow specifying target directory (fixes #101)
closed 2020-09-02 by roleoroleo