roleoroleo/yi-hack-Allwinner #102: RTSP Stream without Video Audio only

schaelrippe, 2020-07-29:
HI! Thanks for this great Project i really appreciate the work wich was going  done here!

Sadly i got a little Problem. I Trying to solve it for hours already.

When i open rtsp://admin:admin@192.168.178.34:554/ch0_0.h264
in VLC Player it doenst show a picture. Just the Sound with arround 2-3 secounds delay.

I tried resetting the cam
looked at the network
put the same link in Synology Survailance station (here i see a video)

info attachment 

[image]

[image]

please help if you can
Thanks!

roleoroleo, 2020-08-03:
Low res works?

schaelrippe, 2020-08-04:
sadly no. in VLC in cant see any video HIres LOWres.

roleoroleo, 2020-11-21:
Try the last version, I fixed a problem.

closed 2020-12-19 by roleoroleo