roleoroleo/yi-hack-Allwinner #104: rRTSPServer not starting with 0.1.8

tbhi, 2020-08-04:
After online upgrade to 0.1.8 no rtsp connections are being accepted.
```
# sh script/wd_rtsp.sh
Failed to create Fifo Source
Failed to create Fifo Source 
Failed to create Fifo Source
```
`/tmp/audio_fifo` does not exist and `RTSP_AUDIO=no`
```
# dmesg
[    3.802075] --->sunxi_daudio_dev_probe<--- end.
[    3.820387] digital_vol:0, lineout_vol:31, main_gain:23, pa_msleep:160, pa_level:1
[    3.820402] adcdrc_cfg:0, adchpf_cfg:1, dacdrc_cfg:1, dachpf:0
[    3.820489] sunxi-internal-codec codec: [sunxi_internal_codec_probe] codec probe finished.
[    3.827702] [sunxi_card_init] card init finished.
[    3.828039] sunxi-codec-machine sndcodec: sun8iw19codec <-> codec mapping ok
[    3.908975] sunxi-codec-machine sndcodec: [sunxi_card_dev_probe] register card finished.
[    3.916639] snddaudio snddaudio0: codec: snd-soc-dummy, codec_dai: snd-soc-dummy-dai.
[    3.917228] snddaudio snddaudio0: snd-soc-dummy-dai <-> daudio0 mapping ok
[    3.975751] [sensor_power_warn] sensor_power_probe: cannot get sensor0_iovdd supply, setting it to NULL!
[    3.975767] [sensor_power_warn] sensor_power_probe: cannot get sensor0_avdd supply, setting it to NULL!
[    3.975777] [sensor_power_warn] sensor_power_probe: cannot get sensor0_dvdd supply, setting it to NULL!
[    4.233321] sunxi-wlan wlan: bus_index: 1
[    4.343622] sunxi-mmc sdc1: sdc set ios:clk 0Hz bm PP pm UP vdd 21 width 1 timing LEGACY(SDR12) dt B
[    4.360888] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    4.381563] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    4.384018] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
[    4.385892] sunxi-mmc sdc1: smc 1 p1 err, cmd 8, RTO !!
[    4.392133] sunxi-mmc sdc1: card claims to support voltages below defined range
[    4.401925] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
[    4.402018] sunxi-mmc sdc1: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
[    4.402178] sunxi-mmc sdc1: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 4 timing SD-HS(SDR25) dt B
[    4.403215] mmc1: new high speed SDIO card at address 0001
[    6.009239] 
[    6.009239] [CPLD_PERIPH] cpld_init ok, [ ver=Dec 27 2019, 09:42:03 ] 
[    6.009708] [CPLD_PERIPH] pwm init ok.
[    6.009718] [CPLD_PERIPH] timer init ok.
[    6.009724] timer resolution:4 MHZ
[    6.009728] [CPLD_PERIPH] HZ = 250 
[    8.588189] [sensor_power_warn] sensor_power_probe: cannot get sensor0_iovdd supply, setting it to NULL!
[    8.588206] [sensor_power_warn] sensor_power_probe: cannot get sensor0_avdd supply, setting it to NULL!
[    8.588215] [sensor_power_warn] sensor_power_probe: cannot get sensor0_dvdd supply, setting it to NULL!
[    9.607675] [CPLD_PERIPH] Timer closed.
[    9.607687] [CPLD_PERIPH] Timer opened.
[    9.929416] cedar_ve: VE real_fre=300000000
[   10.081529] snddaudio snddaudio0: codec_dai set sysclk failed
[   10.081541] snddaudio snddaudio0: codec_dai set set_pll failed.
[   10.081548] snddaudio snddaudio0: codec dai set fmt failed
[   10.081569] snddaudio snddaudio0: codec_dai set clkdiv failed
[   10.151609] s_fmt set width = 1936, height = 1096
[   10.243490] into set sensor hfilp the value:0 
[   10.245022] [VIN]sensor_flip_option:isp0 reset!!!
[   10.245347] into set sensor hfilp the value:0 
[   10.247013] [VIN]sensor_flip_option:isp0 reset!!!
[   10.256338] cedar_ve: VE real_fre=300000000
[   10.258086] cedar_ve: VE real_fre=300000000
[   21.853824] [CPLD_PERIPH] Timer closed.
[   23.307052] cedar_ve: VE real_fre=300000000
[   23.335604] cedar_ve: VE real_fre=300000000
[   23.675459] cedar_ve: VE real_fre=300000000
[   23.870330] snddaudio snddaudio0: codec_dai set sysclk failed
[   23.870343] snddaudio snddaudio0: codec_dai set set_pll failed.
[   23.870351] snddaudio snddaudio0: codec dai set fmt failed
[   23.870375] snddaudio snddaudio0: codec_dai set clkdiv failed
```

roleoroleo, 2020-08-04:
Is "Disable cloud" on or off?

tbhi, 2020-08-04:
on
```HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=no
ONVIF=no
ONVIF_WSDD=yes
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=
FREE_SPACE=0
FTP_UPLOAD=no
FTP_HOST=
FTP_USERNAME=
FTP_PASSWORD=
SSH_PASSWORD=
```

roleoroleo, 2020-08-04:
Probably, there is a bug.
Try to edit /home/yi-hack/script/system.sh
and change
```
if [[ $(get_config RTSP_AUDIO) == "none" ]] ; then
```
to
```
if [[ $(get_config RTSP_AUDIO) == "no" ]] || [[ $(get_config RTSP_AUDIO) == "none" ]] ; then
```

tbhi, 2020-08-04:
Changed that but still the same.  It seems like `rRTSPServer` now won't start unless `/tmp/audio_fifo` exists.  I've changed to `RTSP_AUDIO=yes` and it is now running.

roleoroleo, 2020-08-04:
rRTSPServer should not try to open /tmp/fifo if RTSP_AUDIO=no.

andker87, 2020-08-06:
same here. no way to make the rtsp streaming work, even with with the "new" rRTSPServer" is not working, however with the old if I enable the rtsp audio, then reboot, it works. 

Then after reboot if I go back to the Configuration menu the "rtsp audio" is flagged as "disabled"

v0.1.8

jfallot, 2020-08-08:
The Yi Home app is however still abale to capture the vidéo stream: do you know any other URL that can be used to capture the stream from ZoneMinder?

roleoroleo, 2020-08-09:
Yi app doesn't use rtsp proticol.

rogerb831, 2020-08-09:
I'm having issues also after upgrading. Manuall running wd_rtsp.sh give the following:

`root@xiaoyi:/home/yi-hack/script# ./wd_rtsp.sh
./wd_rtsp.sh: line 90: tail: not found
./wd_rtsp.sh: line 90: rRTSPServer: not found
`

rogerb831, 2020-08-09:
Manual start of rRTSPServer give this error:

> root@xiaoyi:/home/yi-hack/script# /home/yi-hack/bin/rRTSPServer
> Failed to create Fifo Source
> Segmentation fault (core dumped)

roleoroleo, 2020-08-10:
@rogerb831 
Try to replace rRTSPServer with the one posted above.

jfallot, 2020-08-10:
@roleoroleo 
I replaced the rRTSPServer with the one you posted above and changed system.sh according to https://github.com/roleoroleo/yi-hack-Allwinner/issues/104#issuecomment-668630540 (although I have the "Disable Cloud" option set to no)
now the RTSP streaming is working back (I didn't have to reboot the cam)

roleoroleo, 2020-08-11:
Please try with and without cloud disabled.

rogerb831, 2020-08-11:
I've made the change, but I've been running with cloud enabled. It appears the line for cloud enabled already has that OR code. Making the change had no impact for me.

rogerb831, 2020-08-11:
I found the issue (at least in my case). /home/yi-hack/script/wd_rtsp.sh uses "tail" but not through busybox. I changed it to use "/home/yi-hack/bin/busybox tail" and everything just worked.

roleoroleo, 2020-08-12:
/home/yi-hack/usr/bin/tail is a symbolic link to /home/yi-hack/bin/busybox
tail is not in PATH if you run it manually (but it's in PATH when wd_rtsp.sh is started.
If you wanto to run it manually without errors you have to source the environment.
`. /home/yi-hack/script/env`

closed 2020-10-06 by roleoroleo