roleoroleo/yi-hack-Allwinner #105: Yi Dome Guard support

artemy, 2020-08-04:
Camera seems to be Allwinner-based, although markings on the chip state "QG2101B", which is ambiguous to me.

I tried adapting this firmware (with compiling custom images for camera's firmware code (which is r30gb), but that approach didn't work since mtdblock3 partition (home) is not writable.

What would be the steps to try to make the hacks in this repo work for that model?

Attached are mtd table and boot log.
[mtd.txt](https://github.com/roleoroleo/yi-hack-Allwinner/files/5021861/mtd.txt)
[boot.log](https://github.com/roleoroleo/yi-hack-Allwinner/files/5021841/boot.log)

I also have dumped stock firmware, can upload it if needed.

NURIHAKIM, 2020-08-04:
Yep I'm at the same spot!

roleoroleo, 2020-08-04:
The cam is Allwinner based but the structure is different.
The size of the flash is only 8 MB (the y20ga is 16 MB) and the partition schema is different.
I could try to study it if you can share the dump (rootfs and home are enough).

artemy, 2020-08-04:
Both partitions should be in this zip archive: [r30gb_fw.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/5022810/r30gb_fw.zip)

roleoroleo, 2020-08-05:
Ok, I'll look into this.

roleoroleo, 2020-08-05:
As I already noticed, the flash structure is very different.
home is not a jffs2 partition, it's a read only squashfs.
And the size of the flash is too small to add other programs.
Probably it's possible to create a new hack based on SD card but it's a new project.
No chances to integrate this cam in the current project.

roleoroleo, 2020-08-05:
Probably it's possible to save:

- toolchain
- sources
- build environment (partially)

NURIHAKIM, 2020-08-06:
Thanks for looking into it. Going to send some coffee to you!

roleoroleo, 2020-08-06:
Thank you for your contrubution.

roleoroleo, 2020-10-06:
@artemy 
Could you send me your mtdblock4?
Probably I'm able to create a new hack for your cam, the flash content is pretty the same.
See this issue:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/157

artemy, 2020-10-07:
@roleoroleo right I'll try to find time this week

artemy, 2020-10-07:
There it is
[Uploading mtdblock4.bin.zip…]()

roleoroleo, 2020-10-11:
Yes, your cam is r30gb suffix and the hack will be the same as y21ga model:
https://github.com/roleoroleo/yi-hack-Allwinner-v2

roleoroleo, 2020-10-19:
If you want to try this beta, please visit this issue:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/3

Version 0.0.6 contains beta support for r30gb.

janowakps, 2020-11-18:
@roleoroleo what is the situation now? does YI Dome Guard YRS.3019 have alternative firmware?

closed 2020-11-21 by roleoroleo