roleoroleo/yi-hack-Allwinner #107: anyone tried 8.2.0.0A_202007211802 yet?

divadiow, 2020-08-05:
8.2.0.0A_202007211802 has just been released. the release note in the app is "increase the image quality".

Has anyone tried updating a hacked camera and or hacking an already-updated camera?

divadiow, 2020-08-05:
well, saying yes to the update in the app just shows the 2019 version still after a reboot.

I had to remove the hack, update firmware and re-apply the hack. seems OK, though I haven't tested anything but uploading to ftp/yi app and clicking around the yihack pages.

image looks more saturated now perhaps

[image]

spamsool, 2020-08-06:
Have you check the stream bitrate on the new fw? The image bitrate on Allwinner version 2019 it is way lower (like even 3-4 times lower) than in the Mstar version. Mstar model has a way more details shown. You can check the current HD bitrate in the App.

divadiow, 2020-08-06:
my app hovers around 90-160kbps.

I have noticed the file sizes for my ftp uploads are bigger.

bitrate info for 60s vids before and after the firmware update:

[image]

roleoroleo, 2020-08-06:
I will check the new version.
I downloaded it and it's about 4 MB. The previous version was 3.1 MB.
A lot of code...

roleoroleo, 2020-08-06:
I tried it and it seems to work correctly with the hack.
If you update a hacked camera you need to apply again the hack.
Otherwise, you can remove the hack, upgrade the cam and apply again the hack.

spamsool, 2020-08-07:
I updated the fw but rtsp server is not working now. I cannot connect to the stream. How could I check whether is is working fine?
When I try to run it manually:
 root@yi-88fb:/tmp/sd# /bin/sh /home/yi-hack/script/wd_rtsp.sh
/home/yi-hack/script/wd_rtsp.sh: line 90: tail: not found
/home/yi-hack/script/wd_rtsp.sh: line 90: rRTSPServer: not found

roleoroleo, 2020-08-07:
You have to souce the env when you start it manually.
`. /home/yi-hack/script/env.sh`

spamsool, 2020-08-07:
I reset hack to default and reboot the camera - it started to work. Sorry for that, I should have done it first ;)

roleoroleo, 2020-08-07:
No problem.

Dragon0501, 2020-08-08:
Everytime I select the update the camera reboots with the former firmware. How can the firmware be updated? And if it is necessary to remove the hack for it how can the hack be removed?

roleoroleo, 2020-08-09:
When the camera restarts with the same version the reason is "out of memory" during upgrade procedure.

Two possible solution:
1. Unhack the cam before upgrade (see the wiki).
2. Enable swap file.

pbanj, 2020-08-14:
updated my cameras, they seem to be working well.

divadiow, 2020-08-18:
any juicy discoveries about what's changed with the new firmware?

EnricoFerro, 2020-08-24:
For the "Switch on/off the camera" continue the discussion in #120 

For the problem "Baby crying detect". I must continue the test, but I suppose that the problem ( if any ) is in `ipc_cmd' or in mqtt...

roleoroleo, 2020-08-24:
> For the problem "Baby crying detect". I must continue the test, but I suppose that the problem ( if any ) is in `ipc_cmd' or in mqtt...

With ipc_sniff you can check if the system (yi system) detects the event.

EnricoFerro, 2020-08-24:
Ok the ipc_sniff give this result:
```
root@yi-camera:/tmp/sd# ./ipc_sniff
IPC message. Len: 16. Status: No error information!
Parsing message.
04 00 00 00 02 00 00 00 04 60 04 60 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
```

I'm trying changing the header flag and recompile:
```
#define IPC_BABY_CRYING         "\x04\x00\x00\x00\x02\x00\x00\x00\x04\x60\x04\x60\x00\x00\x00\x00"
```

EnricoFerro, 2020-08-24:
Update: 

with this modify, the specific device with :
BSUS*
8.2.0.0A_202007211802

The "baby crying" work correctly.

I've created an artifact: https://github.com/EnricoFerro/yi-hack-Allwinner/suites/1090707352/artifacts/15118350

roleoroleo, 2020-08-24:
But with the old fw 8.2.0.0A_2019* does it work?
If yes, I could add a second define and leave the old one.

EnricoFerro, 2020-08-24:
With another model with 8.2.0.0A_2019* for me work correctly the current value:
```
IPC message. Len: 16. Status: No error information!
Parsing message.
04 00 00 00 02 00 00 00 02 60 02 60 00 00 00 00
```

I've never created a pull request because i suppose that a second define is better.

roleoroleo, 2020-08-24:
Maybe the message is variable?
Maybe it embeds other parameters like intensity, duration, probability...

Aqua-Ye, 2020-09-04:
> When the camera restarts with the same version the reason is "out of memory" during upgrade procedure.
> 
> Two possible solutions:
> 
> 1. Unhack the cam before upgrade (see the wiki).
> 2. Enable swap file.

- could you explain how to enable swap file?
- in case of an unhack, is it possible to save the config in order to restore the hack after upgrading?

pbanj, 2020-09-04:
Yes you can back your config up via the web interface. The swap file toggle is under the configuration tab. I suggest you actually poke around the firmware

noelmom, 2020-09-06:
I can't get my 9FUSY31 running 8.2.0.0A_202007211802 to flash.  The blue light just keeps flashing camera doesn't boot.  Any ideas?

nbendall34, 2020-10-14:
has anyone been able to hack the new fw because I can't get the hack to work on my 9FUS. I just get a flashing blue light constantly.

nbendall34, 2020-10-15:
#165

nbendall34, 2020-10-15:
I have opened a separate issue here for my problem #165

closed 2020-11-21 by roleoroleo