roleoroleo/yi-hack-Allwinner #112: AI Human detection support

BenjaminFaal, 2020-08-10:
https://github.com/roleoroleo/yi-hack-Allwinner/blob/860c007d78289dfb1bd5563f5a5f0bd494503dff/src/ipc_cmd/ipc_cmd/ipc_multiplexer.c#L76

How can i debug IPC messages, id like to add support for AI Human detection.

roleoroleo, 2020-08-11:
Try this sniffer:
[ipc_sniff.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5056969/ipc_sniff.tar.gz)

BenjaminFaal, 2020-08-11:
I receive the following messages, i presume this is in hex format?

```
This seems to be logged when i am visible to the camera
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 15 04 00 00

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 59 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 43 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 32 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 32 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 d2 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 4c 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 0d 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 d4 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 03 04 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 32 04 00 00 

And these when im not visible to the camera
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 fd 03 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 20 03 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 40 03 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 a0 03 00 00 

01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 b2 03 00 00 
```

the 04 and the 03 are "start" "stop" i presume? And what do you think are the values before the 03 and 04?

roleoroleo, 2020-08-12:
Try the reverse test: save the string as a binary file and send it with ipc_cmd.
You should get a notification on the app.

BenjaminFaal, 2020-08-13:
@roleoroleo it turns out the above IPC messages where for something else.

I opened a PR for AI Human Detection here: https://github.com/roleoroleo/yi-hack-Allwinner/pull/114

closed 2020-08-13 by BenjaminFaal