roleoroleo/yi-hack-Allwinner #113: Hostname not working

McGiverGim, 2020-08-12:
Hi! Thanks for this great piece of software! It works incredible, but I have a little problem:
- I'm trying to access the camera by name and not by IP (my router is very limited in this aspect). I have configured the hostname as `camara-entrada` but when it connects to the network it does not appear with this name. The name is empty.

I have to do something more that add the hostname in the configuration page?

roleoroleo, 2020-08-12:
The cam doesn't send the hostname to the dhcp server.
This piece of code is not part of the hack, sorry.
You could try to assign a reservation on your router to have the same ip address at every reboot.

McGiverGim, 2020-08-12:
Ok, understood. The problem is that my router limits to 10 the number of fixed IPs. Maybe is time to add a good router to the system ;)

Thanks for your work!!

closed 2020-08-12 by McGiverGim