roleoroleo/yi-hack-Allwinner #115: ONVIF watermark not working

McGiverGim, 2020-08-13:
Maybe there is something I'm missing, but the `ONVIF watermark` switch does nothing for me.

I use the ONVIF integration in Home Assistant, neither the video or the snapshot are different with this switch enabled/disabled. I suppose it must add the date/time to the image?

roleoroleo, 2020-08-13:
Yes, but only in the snapshot, not in the stream.
Check if ha is using correctly onvif protocol.
If ha doesn't use onvif, it saves the snapshot with ffmpeg through the stream and watermark doesn't appear.

McGiverGim, 2020-08-13:
In theory I'm using Onvif, but I've seen messages that if something goes wrong, it changes to ffmpeg.
Do you know how can I verify?

roleoroleo, 2020-08-13:
Try onvif device manager.

McGiverGim, 2020-08-14:
I think I can see in the little snapshot that Onvif device manager shows the watermark.... so is clear that is a Home Assistant issue. Sorry for spending your time and thanks for your help!
I will try to find the problem.

McGiverGim, 2020-08-17:
@roleoroleo I reopen this again because I've been investigating, and I've found the problem, maybe you want to fix it if possible.

The problem is that the authentication for the snapshot (and in general in the camera) is only of type BASIC and the ONVIF integration in Home Assistant uses DIGEST.

I have made a DIGEST authentication using CURL and I get a 401 unauthorized, but with BASIC it works perfectly. I've verified it too capturing the Chrome requests to the camera interface and I see the authentication used is BASIC too.

Is not a big problem, a PR is on the way to fallback to BASIC in Home Assistant: https://github.com/home-assistant/core/pull/38729

But I think it can be a good idea to support DIGEST, specially because the connection is HTTP only and it will increase a little more the security.

roleoroleo, 2020-08-17:
Which version of HA are you using?
I'm using it every day and lovelace works correctly.

McGiverGim, 2020-08-17:
I'm on the latest one, 114.1, using the ONVIF flow integration. It works, but I don't get the ONVIF snapshot, I only get the ffmpeg snapshot.
Does your hack support DIGEST and maybe I've something configured wrong?

roleoroleo, 2020-08-17:
Ok, I never user onvif with auth enabled.
How did you configure authentication?

McGiverGim, 2020-08-17:
I'm not too sure, I don't remember. Maybe with the official yi app?

roleoroleo, 2020-08-18:
I mean inside home assistant.

McGiverGim, 2020-08-18:
When you execute the onvif flow integration it asks for user and password.

closed 2024-05-13 by github-actions[bot]