roleoroleo/yi-hack-Allwinner #118: BFUS YIcam randomly lose wifi connection

sajikur, 2020-08-18:
on 1.8 firmware, cams (I've 4 BFUS) lose connection and I must to reboot manually to reconnect
Any ideas to solve?
Thanx

roleoroleo, 2020-08-18:
Do you have this problem only with 1.8?

sajikur, 2020-08-18:
Hi, thanx for your answer. No, same problem with 1.7

bonuzzz, 2020-08-18:
weak wifi signal for example

sajikur, 2020-08-19:
No, 4 cams 1 router 2 apn. WiFi signal is good in every rooms

sajikur, 2020-08-19:
News! Searching in configuration Web page of every cam, I see that every 
cam reboots randomly. So, my first report was wrong. They don't loose WiFi 
but reboot randomly. On 1.8 and 1.7 firmware

roleoroleo, 2020-08-19:
Probably the cams are out of memory.
Try to remove unneded services.
Or enable swap file.

sajikur, 2020-08-19:
Could you tell me how to enable swap file?
Unneded services are already disabled.
Thank you for your answer
Mario

sajikur, 2020-08-19:
oops, found the swap file trigger.
I'll let you know if this solved my issue

sajikur, 2020-08-20:
ok, your solution (swap file) seems solve my issue. THANK YOU FOR YOUR SUPPORT!!!

closed 2020-08-20 by sajikur