roleoroleo/yi-hack-Allwinner #119: MQTT retainted messages are retained even the retained option is disabled

sermayoral, 2020-08-18:
Hi everyone.

I have a retained message "baby crying". I have disabled the retain putting a '0' in the option "Retain for baby crying messages". After reinstall the MQTT broker, i receive another retain message.

Is this option working?

roleoroleo, 2020-08-18:
I have to check it.

roleoroleo, 2020-08-18:
I cannot test baby crying remotely.
But i tested motion start, motion stop and file list.
Retain is correctly sent as false when I set it to 0.

sermayoral, 2020-08-18:
Thanks @roleoroleo for testing it.

Maybe i have not cleared the mqtt retained message well. I'm going to check again and share the result.

sermayoral, 2020-08-18:
Yes, it works. I have not deleted the retained messages propertly.

Sorry and thank you!!

closed 2020-08-18 by sermayoral