roleoroleo/yi-hack-Allwinner #122: Possible to change wifi credentials?

Fookes74, 2020-08-19:
Is there any way (aside from re-flashing) to change the SSID credentials stored by the camera? Looking to swap networks and not able to keep same SSID. Thanks.

roleoroleo, 2020-08-19:
Credentials are stored inside a raw partition.

At the moment it's not possible.

Fookes74, 2020-08-20:
Thanks for the response. I take it reflashing the cameras shouldn't be an issue or would you advise against it?

roleoroleo, 2020-08-20:
If you want to change credentials you have to reset the cam with the back button.
If you reflash it, credentials don't change.
The flash partition where they are stored is not erased.

Fookes74, 2020-08-20:
Appreciate you responding. Sorry, 'reset the cam with the back button'? What/where is that?

roleoroleo, 2020-08-20:
https://m.youtube.com/watch?v=-BCTm2VXxkQ

Fookes74, 2020-08-20:
Ahhhh, the reset button. I understand now. I'm presuming the stock firmware is lost though. What is the camera resetting to? The video looks like it's resetting stock firmware. Is this going to be a problem with the flash already being on it?

roleoroleo, 2020-08-20:
The reset procedure restores the factory settings but doesn't change hack.

Fookes74, 2020-08-20:
Right. Sorry, I'm getting confused. Does the hack not overwrite the stock firmware then? If I reset my camera I can simply reconnect it to the new ssid via Yi Home app and the hack will still be there working?

roleoroleo, 2020-08-20:
Yes.
Only the fw upgrade breaks the hack: if you upgrade via Yi app you need to apply again the hack.

closed 2020-10-06 by roleoroleo