roleoroleo/yi-hack-Allwinner #123: how to grab recordings via ssh?

tatoosh, 2020-08-20:
Where can i find the mp4 files in the filesystem by connection with ssh?

Thx a lot

roleoroleo, 2020-08-20:
/tmp/sd/record

tatoosh, 2020-08-20:
Thanks alot!
Is there a possibility also get the screenshots taken by web? `http://<IP>:8080/cgi-bin/snapshot.sh?res=low&watermark=yes`
I want to trigger my doorbell to take a screenshot and send it by push to my mobile.

roleoroleo, 2020-08-20:
imggrabber -r high -w > /tmp/image.jpg

tatoosh, 2020-08-24:
great thanks!! - i changed it a bit, so i can grab it by url `/home/yi-hack/bin/imggrabber -r low -w > /tmp/sd/record/door.jpg`

tatoosh, 2020-08-27:
Can i call a script by URL to renew my file `/tmp/sd/record/door.jpg` ? so this is always my newest jpg?

roleoroleo, 2020-08-27:
It should work.
Call it with the address `http://IP_CAM:8080/cgi-bin/door.sh`

tatoosh, 2020-08-27:
I forgot to add shebang `#!/bin/bash` - now it's working thanks!

tatoosh, 2020-11-10:
Can u please  put this script in the release version?

/home/yi-hack/www/cgi-bin/door.sh
```
#!/bin/bash
/home/yi-hack/bin/imggrabber -r low -w > /tmp/sd/record/door.jpg
```
chmod +x  /home/yi-hack/www/cgi-bin/door.sh

roleoroleo, 2020-11-10:
I don't remember.
Why didn't you use the snapshot web service?
`http://<IP>:8080/cgi-bin/snapshot.sh?res=low&watermark=yes`

tatoosh, 2020-11-10:
With my script i always can write the image screenshot to the same filename `/tmp/sd/record/door.jpg`.
I use this image for my doorbell over home assistant. By a ring a telegram bot sends me the image ;)

roleoroleo, 2020-11-10:
The web service does the same, without saving the image.
It returns directly from the http call.

tatoosh, 2020-11-10:
I need a saved image for my script

tatoosh, 2020-12-04:
Reminder for my question: _Can u please put this script in the release version?_ is that possible for u?

Thx

roleoroleo, 2020-12-04:
Sorry, I forgot.

roleoroleo, 2020-12-04:
I added your feature to the current web service as an option.
The option file create a new image in /tmp/sd/record
So your syntax will be:
```
http://192.168.7.32:8080/cgi-bin/snapshot.sh?res=low&watermark=yes&file=door.jpg
```

https://github.com/roleoroleo/yi-hack-Allwinner/commit/6fdfd0c01bad9bbc6ffea04fff35c2f380abc01f

tatoosh, 2020-12-05:
Thanks a lot, so i can use it with the next release or?

roleoroleo, 2020-12-05:
You can edit this file now copying the commit.
And when I will release the new version you will not lose the changes made.

closed 2020-12-19 by roleoroleo