roleoroleo/yi-hack-Allwinner #124: How to update base firmware

vmendivil, 2020-08-20:
My yi-hack version just got updated to 0.1.9 but I noticed that my base firmware is still on version 8.2.0.0A_201912270941 and there is a new release 8.2.0.0A_202007211802, how can I updated it?

I was thinking on remove yi-hack, update base firmware and then apply the yi-hack again. Is there a better way to do it?

From the yi mobile app, when updating base firmware, it just restarts but doesn't apply the firmware update.

I looked into the wiki but couldn't find much to answer my question.

roleoroleo, 2020-08-20:
See here:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/107

vmendivil, 2020-08-20:
Awesome, that's what I needed. Thank you!

closed 2020-08-20 by vmendivil