roleoroleo/yi-hack-Allwinner #125: Add AI Human Detection

MrPatate87, 2020-08-21:
I love to see the hack developpement growing and really appreciate your work.
Many feature improvement in every version of the hack and all version work fine just now, but I have a qestion abour the new implementation of AI Human Detection.
I use my camera with surveillance station with RTSP, wor fine in HD with audio (great improvement from 0.1.8).
I decide to upgrade to 0.1.9, the update run well and really speed.
I try this new setting and enable AI Human Dectection but I don't really know how this fearure work!
Is it just like originak firmware for the camera recording, or can I made this detection to be use in surveillance station ?
When I try to chhnge the setting of the camera detection in surveillance station, I only can set it to surveillnce station algorythm.....
What is the corrcet use of this new feature??
Thanks again fir all the work and all many update you made for this hack!!!

roleoroleo, 2020-08-21:
AI human detection is a feature of the original firmware.
With the hack you can enable a mqtt message when there is a detection.
So you can integrate it as a sensor on your domotic system.
See the wiki for an example with home assistant.

You can't configure settings about human detection, only enable/disable.

vladeha, 2020-08-28:
After enabling this feature , relevant mqtt topic wasn't created.
Moreover , mqtt motion topic, that was working before, also stopped working .
After disabling AI feature and restarting camera, mqtt motion topic starts working again.

roleoroleo, 2020-08-28:
Is it different if you enable the feature from the yi app?
Does it work?
Or it's the same?

vladeha, 2020-08-28:
moment, I need instal and test it.
I don't have the Yi app

MaxWinterstein, 2020-09-19:
Stumbled across the same when trying to get some motion detection working.

As i was observing the mqtt broker i saw those message:
[image]

roleoroleo, 2020-09-19:
Sorry, I don't know how to solve.
This issue is not related with the hack.
When you enable ai_human_detection, the system doesn't send ipc messages about motion start.

closed 2024-05-13 by github-actions[bot]