roleoroleo/yi-hack-Allwinner #129: show WLAN signal strength

tatoosh, 2020-08-24:
It would be nice to see the WLAN signal strength on the WebGui

elraro, 2020-08-29:
Working on this!

elraro, 2020-08-29:
I would like to know your opinion,

Do you like an icon? text? or both at once?

The text would be: strong, medium, weak and no signal

At the moment it works only with icon

[image]

Best regards

tatoosh, 2020-08-30:
Thanks for asking. I would like to see percentage to optimize the place for the cam or repeater. Image is farly better than nothing ;)
What do you think about placing to the Top Menu?
[image]

elraro, 2020-08-30:
Uhm, I think it's a strange place to set the strength of the wifi signal, when you have a page dedicated to the status of the camera.

@roleoroleo what do you think?

I like the percentage, i will aggregate it

Regards,

roleoroleo, 2020-08-31:
The free space in the top menu is low.
I prefer in the status page, but both icon and percentage if possible.

elraro, 2020-09-20:
[image]

Done!

roleoroleo, 2020-09-26:
@elraro
I have an issue with the value:
https://github.com/roleoroleo/yi-hack-Allwinner/pull/150#issuecomment-699508964

tatoosh, 2020-10-05:
Not working, WLAN is connected but the Status is wrong

[image]

elraro, 2020-10-05:
> 
> 
> Not working, WLAN is connected but the Status is wrong
> 
> [image]

Can you paste the output of /proc/net/wireless ?

tatoosh, 2020-10-06:
After reboot it's fine - here are the results if that helps. Thx

[image]
[image]

elraro, 2020-10-10:
@tatoosh please, if you detect again the problem, run the command before restart. ty!!!

closed 2020-11-21 by roleoroleo