roleoroleo/yi-hack-Allwinner #132: 9FUSY32 with firmware 8.3.0.0A_201912271707, won't flash

Trentmclaughlin, 2020-08-26:
Just got new cameras, they won't flash likley due to the new 8.3 firmware. Anyone working on fixing this. Any way I can help?
When I attempt to flash the orange light comes on and stays on. When I pull out the card, the cameras boot normal and no changes have been made to the card.

roleoroleo, 2020-08-26:
Probably it's a different model, not a different fw.
Try to make a backup using the procedure described in the wiki.

Trentmclaughlin, 2020-08-26:
I just tried the backup (SD card method), same thing. Orange light stays on forever, no changes to the card. 
Model is a 9FUSY32LYS (QR card on back) Model YYS.2518 (bottom of stand) YI Home Camera 3 Date 2020.04 (Back of Box)

roleoroleo, 2020-08-26:
I think you have to use tee serial cable method.

Trentmclaughlin, 2020-08-27:
[9FUSY32LYS.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/5133538/9FUSY32LYS.zip)
Well that wasn't easy for a noob like me. Here is the image dump. I hope you can create/help me with this.

roleoroleo, 2020-08-27:
Do you want to try this hack?
[y25ga_0.1.9.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5136276/y25ga_0.1.9.tar.gz)

Trentmclaughlin, 2020-08-27:
I certainlyappricieate trying to help with this. I tried this new on one two diffrent cards, same thing. continous oragnge light. I tried it a third time with a serial cable attached. Here is the output.
[hachtry2.log](https://github.com/roleoroleo/yi-hack-Allwinner/files/5136834/hachtry2.log)

roleoroleo, 2020-08-27:
```
############## Starting Hack ##############
```

The hack starts but something is wrong in the syntax.
Check if the file config.sh is in linux format with the correct new line char.

Trentmclaughlin, 2020-08-27:
Thats above my knowledge base. Did you write it?

roleoroleo, 2020-08-27:
I would like to understand if the file has changed during the copy.
Open the config.sh file in the microSD and check if the lines end with LF char or CR+LF chars.
For example with Notepad++ (View --> Show Symbol).
https://en.wikipedia.org/wiki/Newline
If the file changed the format and now is in Windows format, it can't run.

Trentmclaughlin, 2020-08-27:
I think you are on to something. I'm not at home with the cameras to test, but looking at the files: I unzipped the files directly to the card using winzip. When I open the files with Notepad++ and show symbols. I see every line ends with CRLF. I then uninstalled Winzip, downloaded 7Zip, and unzipped again and reopened in Notepad++, the lines now end in just LF.

When I get home I will test again and report back.

Trentmclaughlin, 2020-08-28:
Awsome, I'm so Happy, One donation coming your way roleroleo.

ANyone who comes accross here this same problem: Its the file on comment 6 of this thread that worked for me. I went back and tried the official release and it didn't work for me. Not sure why my zip program was converting the file to a diffrent format but it was, and 7zip did once as well. Each time now when I unzip to the SD card I check it in Notepad++ to be sure it only has LF.

closed 2020-08-28 by Trentmclaughlin

roleoroleo, 2020-08-28:
Please check it all features work correctly and send me a feedback.
I will add your model to the supported cams.

Thank for your donation!

Trentmclaughlin, 2020-08-31:
Everything works. When I goto the snapshot page, the camera takes a snap then seems to go through a reboot or reconnection,  but that is the only thing I have noticed.  

I cant get home assistant to see ONVIF, but I think that is a network issue on my side. I have added the cameras to motioneye inside home assistant.

roleoroleo, 2020-08-31:
The reboot is a known issue.
Try to enable swap file.

To check onvif, use onvif device manager and try to connect on the same subnet.