roleoroleo/yi-hack-Allwinner #133: .ssh/authorized_keys Don't work correctly as descripted in the wiki

EnricoFerro, 2020-08-27:
I've seen that the documentation on the page [wiki/SSH:-login-with-keys ](https://github.com/roleoroleo/yi-hack-Allwinner/wiki/SSH:-login-with-keys) don't work correctly.

The problem derive from a different root path in `/etc/passwd` between 
https://github.com/TheCrypt0/yi-hack-v4/blob/master/src/static/static/rootfs/etc/passwd
and yi-hack-allwinner.

I've resolved with a small workaround in  [src/static/static/home/yi-hack/script/system.sh](https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/src/static/static/home/yi-hack/script/system.sh) adding this line:
```
    sed -i 's|/root|/home/yi-hack-v4|g' "/home/yi-hack/etc/passwd"
```

With this workaround the root path of root user change from `/root` to `/home/yi-hack-v4/' when `ssh root@IP-CAM` is called

closed 2020-08-28 by roleoroleo

dentex, 2020-09-27:
Hello,
sorry to bother you writing below a closed issue... but I coundn't use the authorized key to login in any way.

So I manually edited `/home/yi-hack/etc/passwd` changing    

`root:$1$0K96xxxxxxxxxxxxxxxxxxxxxxx/Nr0:0:0:root:/root:/bin/ash`    

to    

`root:$1$0K96xxxxxxxxxxxxxxxxxxxxxxx/Nr0:0:0:root:/home/yi-hack:/bin/ash`

and then I've been able to login with the ssh key.

I'm using
> Firmware Version | 0.1.9
> Base Version | 8.2.0.0A_202007211802

roleoroleo, 2020-09-27:
Yes, I know.
I have to release the new 0.2.0.
The current 0.1.9 doesn't contain the merge above.

dentex, 2020-09-27:
ohh... OK. 
I don't know why I assumed this fix was already into the current release.
Thanks.

PR #134 (2020-08-28): Fix for mapping the root path for .ssh/authorized_keys ( #133 )
This modify change the root path from `/root` to `/home/yi-hack-v4/` and allow to use the `.ssh/authorized_keys` as descripted in the documentation [wiki/SSH:-login-with-keys ](https://github.com/roleoroleo/yi-hack-Allwinner/wiki/SSH:-login-with-keys)

( Fix #133 )