roleoroleo/yi-hack-Allwinner #135: Enhance Baby Crying Command

EnricoFerro, 2020-08-28:
This post for add additiona feature in Camera Setting for Baby Crying. 

As ask me by @roleoroleo last week in #107 :
>Maybe the message is variable?
Maybe it embeds other parameters like intensity, duration, probability...

I make some additional experiment with ipc_sniffer and original software and configuration for Baby Crying and in the original configuration there is the decibel configuration for the volume.

There are this codes for setting founde for the volume:
```
#define IPC_BABYCRYING_50  "\x02\x00\x00\x00\x08\x00\x00\x00\x3b\x10\x01\x00\x04\x00\x00\x00\x32\x00\x00\x00"
#define IPC_BABYCRYING_60  "\x02\x00\x00\x00\x08\x00\x00\x00\x3b\x10\x01\x00\x04\x00\x00\x00\x3c\x00\x00\x00"
#define IPC_BABYCRYING_70  "\x02\x00\x00\x00\x08\x00\x00\x00\x3b\x10\x01\x00\x04\x00\x00\x00\x46\x00\x00\x00"
#define IPC_BABYCRYING_80  "\x02\x00\x00\x00\x08\x00\x00\x00\x3b\x10\x01\x00\x04\x00\x00\x00\x50\x00\x00\x00"
#define IPC_BABYCRYING_90  "\x02\x00\x00\x00\x08\x00\x00\x00\x3b\x10\x01\x00\x04\x00\x00\x00\x5a\x00\x00\x00"
```

And I see also these codes for activate and deactivate the sound alert:
02 00 00 00 08 00 00 00 3a 10 01 00 04 00 00 00 01 00 00 00
02 00 00 00 08 00 00 00 3a 10 01 00 04 00 00 00 02 00 00 00

This codes is different for the one in the in the [ipc_cmd.h](https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/src/ipc_cmd/ipc_cmd/ipc_cmd.h), but the codes in ipc_cmd.h works correctly.

In the next days, I want make same test for adding these flags in the camera setting page, and in ipc_cmd.

EnricoFerro, 2020-08-28:
I've checked for:
```
#define IPC_BABY_CRYING_2       "\x04\x00\x00\x00\x02\x00\x00\x00\x04\x60\x04\x60\x00\x00\x00\x00"
```
and this flag is the same for all change that I've made with the standard app.

I will check for the other flag in the next days( with the other camera):
```
#define IPC_BABY_CRYING         "\x04\x00\x00\x00\x02\x00\x00\x00\x02\x60\x02\x60\x00\x00\x00\x00"
```

roleoroleo, 2020-12-27:
Code refactoring in this commit:
https://github.com/roleoroleo/yi-hack-Allwinner/commit/18edd2eb35d584a27e6a2cf03a8fb8c55f6eba35

closed 2021-02-15 by roleoroleo