roleoroleo/yi-hack-Allwinner #136: Record duration

scmanjarrez, 2020-08-28:
Hi, thank you very much for your work. It works really well.

Is it possible to modify the video record duration? I haven't found any parameter related to that in the web interface.

Thanks.

roleoroleo, 2020-08-28:
No sorry.
This feature is implemented by yi.
It's not under my control.

scmanjarrez, 2020-08-28:
Ok, thank you!

closed 2020-08-28 by scmanjarrez

ricardojlrufino, 2021-03-23:
Its possible to start a record video like 

root@xiaoyi:/home/yi-hack/bin# ./h264grabber -r HIGH -d > video.mp4

- Ajust VLC Codec for H.264
https://www.youtube.com/watch?v=Vk5viMViSXk

ricardojlrufino, 2021-03-23:
One options is change source of h264grabber, to set a timer ...

ricardojlrufino, 2021-03-23:
You can use:
./h264grabber -r HIGH -d > /tmp/sd/record/VIDEO_1.MP4  & sleep 30 ; kill $!

roleoroleo, 2021-03-23:
The output format is not exactly an MP4 file.
It's a h264 raw stream, but you can read or convert it easily.
And with this method, audio is not available.

ricardojlrufino, 2021-03-23:
truth I realized this now ...
How do I get the audio and video ?

roleoroleo, 2021-03-23:
Sorry, is not possible.