roleoroleo/yi-hack-Allwinner #139: Home Assistant Integration with the newest version not working!

colethegamer123, 2020-09-02:
Hello Dear Roleoroleo,

First of all Thank you very much for your fantastic firmware. I was able to use it as a Generic IP Cam in ONVIF.

I would like to use Home Assistant Yi Home Integration, but this is not working with this version.

Can you please maybe check my community thread about this, and see the informations I collected?

https://community.home-assistant.io/t/yi-home-1080p-integration/223742/2

Thank you very much,

Best regards,

roleoroleo, 2020-09-02:
Yi integration is made for old camera models.
Please check the wiki.

colethegamer123, 2020-09-02:
I see now okey, so that means from Home Assistant its not possible to use Microphone and Speaker modes like in Yi app.

Thank you.

roleoroleo, 2020-09-02:
It's not possible.

colethegamer123, 2020-09-02:
Thank you very much for the really fast response. Have a nice day. I close with comment. Best regards,

closed 2020-09-02 by colethegamer123