roleoroleo/yi-hack-Allwinner #140: Fix for deleting old record

EnricoFerro, 2020-09-04:
Hi,

I've a  dubt that the cron job `clean_record.sh` don't seems to find the correct directory:
https://github.com/roleoroleo/yi-hack-Allwinner/blob/318db69195255cd496360c8e2c480a48064cea7a/src/static/static/home/yi-hack/script/clean_records.sh#L31

I think that this line work better:
```
ls -lt /tmp/sd/record | /home/yi-hack/usr/bin/tail -n1 | awk '{print $9}'
```

roleoroleo, 2020-09-04:
I don't know if crond get the environment from system.sh.
No problem to apply your patch.

EnricoFerro, 2020-09-04:
I make some check... ..I need some time to create a little bit of spam in the sd.

The real problem is that the `/home/yi-hack/usr/bin` is not catch in ssh shell. 
I suppose that the same problem is in the crond

roleoroleo, 2020-09-04:
You could check the path env in /proc.

roleoroleo, 2020-09-04:
The path used by crond seems to be ok:
```
PATH=/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/base:/home/base/tools:/home/yi-hack/bin:/home/yi-hack/sbin:/home/yi-hack/usr/bin:/home/yi-hack/usr/sbin:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbin
```

EnricoFerro, 2020-09-04:
Ok, I've verified. My fault.

Close the issue

closed 2020-09-04 by EnricoFerro