roleoroleo/yi-hack-Allwinner #141: 9FUSY31 running 8.2.0.0A_202007211802 not flashing.

noelmom, 2020-09-06:
Can't seem to flash a 9FUSY31 running 8.2.0.0A_202007211802 when I power on the camera it goes from amber to then flashing blue constantly.  What can I do to further troubleshoot?

scruff-01, 2020-09-06:
Try formatting the 16GB or less SD card in the camera first.

noelmom, 2020-09-06:
Thanks for the tip, however, that didn't resolve the issue.  I was able to enable telnet and tried running the script but getting a syntax error (screenshot below).  Will try running a previous release.

[image]

scruff-01, 2020-09-06:
Have you tried both flashing the tar.gz and flashing with the tar.gz extracted? Maybe try an older version 15 or 16 first

noelmom, 2020-09-06:
My issue was the SD card...  Purchased a Sandisk 16GB and worked without any issues.  Thanks for the help.

closed 2020-09-06 by noelmom

scruff-01, 2020-09-06:
Ah yes. That seems to be a recurring problem with cams these days.