roleoroleo/yi-hack-Allwinner #142: BFUS31* with 8.2.0.0A_201911270941 FW not properly updating

ruohki, 2020-09-06:
Hey,
i applied the patch to my camera but it seems that partial parts are not working corectly. Things seem to get updated. Folders get renamed on the sd card and the camera restarts. But after the restart nothing really changes. The Yi services are still running

```
root@xiaoyi:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  179 root         0 SW   [oom_reaper]
  180 root         0 SW<  [writeback]
  182 root         0 SW   [kcompactd0]
  183 root         0 SW<  [crypto]
  184 root         0 SW<  [bioset]
  186 root         0 SW<  [kblockd]
  224 root         0 SW   [sys_user]
  231 root         0 SW   [kworker/0:1]
  232 root         0 SW<  [cfg80211]
  239 root         0 SW<  [watchdogd]
  253 root         0 SW<  [spi0]
  263 root         0 SW   [kswapd0]
  348 root         0 SW<  [bioset]
  353 root         0 SW<  [bioset]
  358 root         0 SW<  [bioset]
  363 root         0 SW<  [bioset]
  368 root         0 SW<  [bioset]
  373 root         0 SW<  [bioset]
  378 root         0 SW<  [bioset]
  383 root         0 SW<  [bioset]
  402 root         0 SW   [irq/302-sunxi-m]
  404 root         0 SW   [irq/166-sdc0 cd]
  405 root         0 SW   [irq/303-sunxi-m]
  430 root         0 SW   [kworker/u2:2]
  463 root         0 SW<  [kworker/0:1H]
  466 root         0 SW<  [bioset]
  467 root         0 SW   [mmcqd/0]
  482 root         0 SWN  [jffs2_gcd_mtd3]
  487 root       980 S    -/bin/sh
  530 root         0 SW   [ksdioirqd/mmc1]
  533 root         0 SW   [RTW_XMIT_THREAD]
  534 root         0 SW   [RTW_CMD_THREAD]
  535 root         0 SW   [RTWHALXT]
  536 root       680 S    ./log_server
  537 root       984 S    ./dispatch
  540 root       980 S    /usr/sbin/telnetd
  958 root     39828 S    ./rmm
 1046 root      1400 S    /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var
 1050 root      3040 S    ./mp4record
 1051 root      1116 S    ./cloud
 1052 root      3824 S    ./p2p_tnp
 1053 root      3812 S    ./oss
 1054 root       676 S    ./watch_process
 1091 root       844 S    pure-ftpd (SERVER)
 1105 root       876 S    dropbear -R -B
 1107 root       668 S    ipc_multiplexer
 1150 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.script
 1232 root       996 S    {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 1275 root      2072 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type tdn:Netw
 1883 root       904 R    dropbear -R -B
 1900 root       980 S    -ash
 4975 root         0 SW   [kworker/0:2]
20612 root         0 SW   [kworker/u2:3]
23546 root         0 SW   [kworker/u2:0]
23553 root         0 SW   [kworker/u2:1]
24515 root       980 S    sleep 10
24528 root       980 R    ps
```

but the webserver hosting the dashboard and other things are not running. What am i missing here?
I formated the card to fat32 (4gb partition), copied the files to the root of the card (yes i extracted the tar.gz). I did setup a 100% fresh camera using the yi home app and then applied the patch. Let it rest for around 15 minutes (constant blue light)

0none, 2020-09-11:
How large is the SD Card used?
it seems some newer cameras have problems using anything other/larger than 16GB, even with a smaller partition.

ruohki, 2024-04-15:
Meh completely forgot about this. Its a 16gb card. But i think this issue can be closed.

closed 2024-08-16 by github-actions[bot]