roleoroleo/yi-hack-Allwinner #143: change or partially disable MQTT messages for motion video files

roobbb, 2020-09-08:
Hello.

I really like this hack and I love to discover/ use its possibilities. Excellent work.

I have a question regarding MQTT messages. When I receive messages because of motion detected, these messages contain brackets -->[ ]. 
e.g. 
`
{"yicam/motion_detection":"motion_start","yicam/motion_files":"{\u000a\u0022start\u0022:\u00222020-09-08T14:15:15+0200\u0022,\u000a\u0022end\u0022:\u00222020-09-08T14:14:22+0200\u0022,\u000a\u0022files\u0022:[ \u00222020Y09M08D14H/14M00S60.mp4\u0022 ]\u000a}\u000a"}
`

My MQTT-Server doesn´t like closing brackets inside messages and throws an error each time a message comes in
`
Error parsing >]< for files
`

Where and how could I avoid these brackets around the video file strings? Or could I alternatively disable the video files part while keeping the rest?

Thank you very much and kind regads
roobbb

CAM: Yi home 1080p BFUSY31
Hack: y20ga - v0.1.9

roleoroleo, 2020-09-08:
Brackets are the standard notation for array in json format.
They cannot be removed.

roobbb, 2020-09-09:
Hello.

Thank you for your quick reply. I thought my problem is simply a bracket inside the message. But it seems now a bit different.

I receive usually a normal json array like this
`
2020.09.09 11:19:44 yicam/motion_files:{
"start":"2020-09-09T11:19:36+0200",
"end":"2020-09-09T11:18:34+0200",
"files":[ "2020Y09M09D11H/18M00S60.mp4" ]
}
`

or a like that
`
2020.09.09 11:21:46 yicam/motion_files:{
"start":"2020-09-09T11:17:33+0200",
"end":"2020-09-09T11:20:36+0200",
"files":[ "2020Y09M09D11H/20M00S60.mp4", "2020Y09M09D11H/19M38S22.mp4", "2020Y09M09D11H/18M00S60.mp4", "2020Y09M09D11H/17M34S26.mp4", "2020Y09M09D11H/16M00S60.mp4" ]
}
`

Everything is fine with these messages. But sometimes I got such messages:
`
2020.09.09 11:16:06 yicam/motion_files:{
"start":"2020-09-09T11:13:56+0200",
"end":"2020-09-09T11:16:02+0200",
"files": ]
}
`
`
2020.09.09 11:17:12 yicam/motion_files:{
"start":"1970-01-01T01:00:00+0100",
"end":"1970-01-01T01:00:00+0100",
"files": ]
}
`

"files": ] looks like an broken array. The start or end date look even strange (1970?). Do you have any hint how could I avoid such behavior?

NTP is turned on and points to my router. Any wan traffic from or to yi-cam is blocked.

Thank you and best regards
roobbb

roleoroleo, 2020-09-09:
I will check if there is a problem with these message.

roleoroleo, 2020-09-09:
Ok, I solved the issue. Now it returns an empty array: "[ ]".
Next release.

roobbb, 2020-09-09:
That was pretty fast. Thank you very much.

Kind regards
roobbb

closed 2020-10-01 by roobbb