roleoroleo/yi-hack-Allwinner #145: FTP Upload Path

mpuff, 2020-09-08:
Hello,

is it possible to add a upload directory path for ftp?

Kind regards

roleoroleo, 2020-09-08:
See this commit:
https://github.com/roleoroleo/yi-hack-Allwinner/commit/318db69195255cd496360c8e2c480a48064cea7a

mpuff, 2020-09-10:
I'm looking forward to the next release

danielk117, 2020-09-21:
@roleoroleo Is there an ETA for the next release?

mpuff, 2020-09-21:
that would be great with the ftp upload path

roleoroleo, 2020-09-21:
No ETA at the moment.
I have to find the time to test the new image.

danielk117, 2020-09-24:
Do you have a built image with the latest changes? I can help you with testing.

roleoroleo, 2020-09-24:
Yes:
[y20ga_0.2.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5278673/y20ga_0.2.0.tar.gz)
Rename it from tar.gz to .tgz

mpuff, 2020-09-25:
i have try now to add a FTP-Directory, i have add **/Kamera/yi-home/**, than i click on save and reboot the camera, than the FTP-Directory box is emty.

danielk117, 2020-09-25:
the new mqtt messages does work. 👍 
what about #150 ?
next i will have a look at the new ftp directory.

roleoroleo, 2020-09-26:
> if mqtt "AI Human Detection" is enabled, `motion_detection=motion_start` and `ai_human_detection=human_start` or `ai_human_detection=human_stop` doesnt work.
> if mqtt "Baby crying detect" is enabled, `motion_detection=motion_start` and `baby_crying=crying` does work again.

This is a know issue of the yi firmware.

> video recording does not work for me. neither on sd-card nor ftp.
> cloud is disabled. "Recording without Cloud" and "Save video when a motion is detected" are enabled...
> where can i find log files? /var/log is empty...

If "AI Human Detection" is enabled, probably it doesn't work.
Otherwise it must work.

> what about #150 ?

Merged and ready for next release.

danielk117, 2020-09-28:
I didn‘t know if the human_detection was enabled when ftp fails. 
I will give a new try the next days.

mpuff, 2020-09-30:
ftp upload is working, but only when i chage i set the path in the system.conf.
FTP_DIR=/Kamera/yi-home

when i set it up in the gui, the FTP_DIR is emty, also in the file.

Catfriend1, 2020-09-30:
@mpuff please try to clear the browser cache.

mpuff, 2020-09-30:
thank you for this info, i have now opened it with another browser and the ftp path is in the box.

closed 2020-09-30 by mpuff