roleoroleo/yi-hack-Allwinner #146: firmware 8.3.0.0A_201912271707

larsvandoremalen, 2020-09-13:
Just tried to use this on a 9FUSY32KOU7XL7200319 with firmware  8.3.0.0A_201912271707 but the connection is refused when trying to connect to the camera IP. Tried to downgrade to 8.2 firmware and thought I bricked it for a moment but using the unhack archive it is back to stock. 

Does anybody  have any pointers on how to make this work on 8.3 firmware or did I just do something wrong?

Thanks!

roleoroleo, 2020-09-13:
I think 8.3.0 isn't a new fw but a different cam.

larsvandoremalen, 2020-09-13:
ha anyway to find that out for sure and/or which chipset is being used, without a serial connection? Otherwise I should have a ttl32 lying around somewhere but will have to search a few boxes. Thanks for thelp, much obliged!

roleoroleo, 2020-09-14:
You could try if this procedure works:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card)

larsvandoremalen, 2020-09-14:
yeah that works, did that directly after I recovered. You can find the files here (except mtdblock7 which is a 0kb file ): https://filebin.net/msnx5yok8b0tz624

roleoroleo, 2020-09-14:
Your cam is a y25ga.
The support is experimental, committed but not released.
https://github.com/roleoroleo/yi-hack-Allwinner/commit/4056f62afcc1060bfdfcb84ff8f484d7056111c0

larsvandoremalen, 2020-09-14:
That is really great, shall I wait for release or would you like me to test a dev version?

roleoroleo, 2020-09-14:
Here you can find a dev version:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/132#issuecomment-681915209

larsvandoremalen, 2020-09-15:
ok  I have flashed the dev version from the other thread. Still refuses web connections although I am able to connect a telnet client and run pings on it. Anything I can do to help resolve?

roleoroleo, 2020-09-15:
Post a `ps` command and `la -laR /` command.

larsvandoremalen, 2020-09-15:
sure see txt file attached, the ls command was too long to paste here. 
[commands.txt](https://github.com/roleoroleo/yi-hack-Allwinner/files/5224477/commands.txt)

roleoroleo, 2020-09-15:
Which archive did you use to apply the hack?

larsvandoremalen, 2020-09-15:
I used the file you posted in issue 132 and copied all the contents of the tar to the SD file: 
[image]

roleoroleo, 2020-09-16:
The hack is ok but the archive is not installed.

larsvandoremalen, 2020-09-16:
Ok did that and the light is on orange for a few seconds and it does a restart, connects to wifi properly (flashing blue then continuous blue)(log is written to sd card) but it still refuses connections to the ip adress from either pc or the yi app for that matter.

jempo, 2020-09-25:
Hi All, i've the same issue for the same FW. Does anyone have an update about this model?
Many Thanks

roleoroleo, 2020-09-26:
I need to check again the update procedure.
I think there is a problem with the name of the file.

mleon0304, 2020-09-27:
Hello!

Can you describe how you're trying to connect to your camera?? e.g. http://192.168.3.44         ???

jempo, 2020-09-27:
Hi @mleon0304, sure. It's the same method described from @larsvandoremalen in previous post and in the README of the project.
I'm trying to connect to http://IP:8080 or with the Yi Application after the hack, but both are unresponsive with the hack. Reverting the hack with the procedure [Unbrick the hack](https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam) all return normal and camera connect to Yi Application.

roleoroleo, 2020-09-27:
After the upgrade, what's the extension of the file home_y25gam?
stage? done? Or nothing?

jempo, 2020-09-27:
Hi @roleoroleo, i'm using the archive posted in this [thread](https://github.com/roleoroleo/yi-hack-Allwinner/issues/132#issuecomment-681915209).
Th archive have `home_y25gam.stage` file, after flash the file was without extension.
In the screenshot the situation after flash. 
Do you need more info?

[image]

roleoroleo, 2020-09-27:
If the extension isn't .done the upgrade process is not completed successfully.

roleoroleo, 2020-09-27:
Telnet works. It's true.
Try to run ls and post the output.
```
ls -laR
```

jempo, 2020-09-28:
Yes, telnet works. The LS command of `/root` folder is this.

`
drwxrwxr-x    2 root     root            30 Nov 25  2019 .
drwxrwxr-x   14 root     root           195 Nov 25  2019 ..
-rw-rw-r--    1 root     root             0 Nov 25  2019 topfile
`
Did you need about other folder?

roleoroleo, 2020-09-28:
```
ls -laR /
```

jempo, 2020-09-28:
Thanks :) ok here in the attachment the output.
[ls.log](https://github.com/roleoroleo/yi-hack-Allwinner/files/5290638/ls.log)

roleoroleo, 2020-09-28:
I checked your file system and I can confirm that the hack works but the update procedure doesn't start.
I don't know how to fix the problem without a cam.

jempo, 2020-09-28:
Ok @roleorole, thanks for your help! I don't know if there is a debug procedure when camera flashing the hack. Maybe with more logs we can find the problem.

roleoroleo, 2020-09-28:
Yes.
We could try to add some log to the /home/app/init.sh file where there the procedure starts:
```
...
elif [ -f /tmp/sd/home_y25gam ]; then
    echo "---tmp/sd/home_y25gam exist, update begin---"
    dd if=/tmp/sd/home_y25gam of=/tmp/newver bs=22 count=1
    newver=$(cat /tmp/newver)
...
```
And to the file /home/base/tools/extpkg.sh

jempo, 2020-09-28:
Ok but i don't have understand where are the `/home/app/init.sh`. Also, what we need to log? and were are placed the logs?

jempo, 2020-09-28:
ok thanks, but i see that in the init.sh in tha camera there is only mention about the model `home_y20gam` and not the `home_y25gam`, it's normal?

i post the init.sh from the camera

[init.script.log](https://github.com/roleoroleo/yi-hack-Allwinner/files/5294008/init.script.log)

roleoroleo, 2020-09-28:
Are you sure that you have a 8.3.0 firmware?
You should have home_y25gam label.

jempo, 2020-09-29:
Yeah i'm sure. I've do right now the unhack and checked in the YI Application.
It could be that the unhack change the init.sh? Because it's strange also for me that there is the home_y20gam in my init.sh if the model is y25..
[image]

roleoroleo, 2020-09-29:
Yes, if you used the released unhack procedure, now you have the wrong init.sh.
The unhack procedure is only for y20 model, at the moment.
I will create a new unhack for y25 model and I will send you.

jempo, 2020-09-29:
Ok many thanks, i wait for the new unhack :)

roleoroleo, 2020-09-29:
They are almost the same.

Here it is:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack

jempo, 2020-09-29:
Ok the new unhack works, now returning to the logs, there is some preferred point in the init.sh that can i put the log to see what goes wrong in the update?

roleoroleo, 2020-09-29:
I think update should work now.

jempo, 2020-09-29:
Ohh yeah! it works! Many Thanks @roleoroleo 🥇  
I don't know what is your release schedule, but the y25 hack works for me now :)

[image]

aidalover, 2020-10-03:
> I think update should work now.

hi, i also have a 9FUSY32 with firmware 8.3.0.0A_201912271707. where can i find a link to the working hack or are you gonna start it in a new project? thanks....

roleoroleo, 2020-10-03:
In the release page.

closed 2020-11-21 by roleoroleo

ykhandler, 2020-12-20:
> yeah that works, did that directly after I recovered. You can find the files here (except mtdblock7 which is a 0kb file ): https://filebin.net/msnx5yok8b0tz624

@larsvandoremalen, can you please send me the mtdblock0 to mtdblock4, i got 1 camera stuck. Thanks.