roleoroleo/yi-hack-Allwinner #147: Camera cannot connect to Wifi

sermayoral, 2020-09-14:
I hope anyone can help me.

My camera was working well, when today, suddently, when i have tried to connect to my camera, it was not connected to Wifi.

After reboot it, the camera still cannot connect to wifi. I had the yi home app disable, so i decided to remove the hack in order to reset the camera to start the process again. When I turn on the camera, it says "waiting to connect", but after following the process, the camera give me an wifi error and it cannot connect to Wifi.

The last thing i did was https://github.com/roleoroleo/yi-hack-Allwinner/issues/144. Maybe i make a mistake, maybe i have another problem, Im not sure, but I a couple of days of change the hostname, the camera lost the wifi access.

Please, could someone give me an advice of how to proceed to solve this problem? Thanks!!

roleoroleo, 2020-09-15:
Probably there is something wrong in the file:
/home/app/script/wifidhcp.sh 

And actually the unhack procedure doesn't restore the original wifidhcp.sh (because it's not released yet).

sermayoral, 2020-09-15:
It works @roleoroleo 

Thank you very much :-)

closed 2020-09-15 by sermayoral