roleoroleo/yi-hack-Allwinner #148: Unable to watch videos on the sd card through the yi app

GrandeSmith, 2020-09-18:
Tried to watch all the recoded clips on the sd card through the yi app with no luck: every time i try to move the timeline , the cursor move back to the first clip.

roleoroleo, 2020-09-19:
Sorry but this feature is not related to my hack.
I don't know how to help you.

GrandeSmith, 2020-09-19:
Hi,

The yi app works properly without the hack, as soon as I install the hack I
get this strange behaviour...

ferinelo, 2020-09-24:
@GrandeSmith, have you set "Timezone" on "Configurations" web UI page? If so, remove the value and try again - I have faced similar issue related to timezone setting.

GrandeSmith, 2020-09-26:
@ferinelo Man, you made my day! With the custom firmware if you change the Timezone you get that strange behaviour. Removing the Timezone now i can go back and forth through the recorded videos correctly. The drawback is that videos are recorded in a folder named after the timestamp Minus 2 hours:-( @roleoroleo I'm wondering how the hack can lead ti this behaviour...

roleoroleo, 2020-09-26:
Now I remember.
https://github.com/roleoroleo/yi-hack-MStar/issues/167
Probably the yi firmware doesn't use correctly the time function (mktime, gmtime, etc...).

ferinelo, 2020-09-26:
Also, without timezone being set, a watermark shows wrong hour (-2h) on a snapshot. 
It looks like we need to live with that. Moving the timeline in Yi app is more important to me.

closed 2020-09-27 by GrandeSmith