roleoroleo/yi-hack-Allwinner #151: ftppush.sh failing to upload with correct username, password, and host address

CaptianNemo, 2020-09-22:
I set up a camera to upload to ftp server. Entered username, password, and host. Double checked they are correct in the system.conf file. Manually executed ftpput with correct information, copied test file to server successfully. ftppush.log is full of failed attempts. 

Example: https://pastebin.com/X9ZjLPLQ

There must be a problem with the way the script gets the username, password, or host address from the system.conf file, else the manual use of ftpput would also fail, correct?

roleoroleo, 2020-09-22:
Enter with ssh and run it manually.
You will see the errors in console.

Try a plain text password.

CaptianNemo, 2020-09-25:
I copied and modified ftppush.sh to put the FTP information into a text file. Here's the output (login, password, and ip changed for security reasons):
[FTP Login] -p [FTP Password] [FTP Address] /38M00S60.mp4 /tmp/sd/record/2020Y09M24D23H/38M00S60.mp4
[FTP Login] -p [FTP Password] [FTP Address] /39M00S60.mp4 /tmp/sd/record/2020Y09M24D23H/39M00S60.mp4
[FTP Login] -p [FTP Password] [FTP Address] /40M00S60.mp4 /tmp/sd/record/2020Y09M24D23H/40M00S60.mp4
[FTP Login] -p [FTP Password] [FTP Address] /41M00S60.mp4 /tmp/sd/record/2020Y09M24D23H/41M00S60.mp4
[FTP Login] -p [FTP Password] [FTP Address] /42M00S60.mp4 /tmp/sd/record/2020Y09M24D23H/42M00S60.mp4

I changed line 148 of ftppush.sh from:
"if ( ! ftpput -u "${FTP_USERNAME}" -p "${FTP_PASSWORD}" "${FTP_HOST}" "/$(lbasename "${UTF_FULLFN}")" "${UTF_FULLFN}" ); then"
To:
"if ( ! ftpput -u "${FTP_USERNAME}" -p "${FTP_PASSWORD}" "${FTP_HOST}" "$(lbasename "${UTF_FULLFN}")" "${UTF_FULLFN}" ); then"

Now it copies the files successfully. Is there a way to make it copy the full YMDHMS name of the folder/file? Instead of "42M00S60.mp4" have it be "2020Y09M24D23H42M00S60.mp4"? That way it never overwrites files?

roleoroleo, 2020-09-26:
Ok, thank you for your contribution.
See this for the last question: https://github.com/roleoroleo/yi-hack-Allwinner/commit/7391e0a2bd4495d1a8dc78820adb3c3610832399

closed 2020-11-21 by roleoroleo