roleoroleo/yi-hack-Allwinner #152: How to disable video recording?

Garywoo, 2020-09-24:
I noticed that my camera is recording motion events to the SD card. I want to stop videos being recorded.

In the Yi Hack settings, I found 'Save video when a motion is detected', however the description states:

> If enabled, video will be saved only when a motion is detected.  
> If disabled, video will be always saved.  
> (It takes effect only if recording is enabled)

If I disable this setting, I'm expecting that the camera will still record video to the SD card. The last sentence hints that recording can be disabled, but I cannot find a setting to control this. I just want to stream video without saving it.

In the Yi app 'Activity Detection Recording' is off.

How can recording be disabled?

pbanj, 2020-09-29:
just disable the cloud and dont turn recording on, it is that simple.

https://cdn.discordapp.com/attachments/399670410565910529/760538069261746216/unknown.png

Garywoo, 2020-09-29:
I'd still like to use the Yi app to view the camera, but I don't want to record.

lukakiro, 2020-10-22:
Disabling the cloud, is it still possible to see the streaming from the YI app?

roleoroleo, 2020-10-22:
> I'd still like to use the Yi app to view the camera, but I don't want to record.

I don't remember if it's possible to disable recording from the app.

> Disabling the cloud, is it still possible to see the streaming from the YI app?

No.

closed 2020-12-27 by roleoroleo