roleoroleo/yi-hack-Allwinner #154: Add to YiHome after Yi hack

colethegamer123, 2020-09-29:
Hello Mr.
I have Yi Home 1080p Cameras, one I added to Yi Home and after installed Yi-hack firmware, this can be viewed by Yi Home app and HomeAssistant too but 

the other what I did in the other way and installed firmware first and after tried to add to Yi Home, but I cannot. There is any solution to add to Yi Home AFTER Yi-hack firmware? 

Thank you!

ferinelo, 2020-10-01:
Have you tried:
- un-hack: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack
- add the camera to the app
- hack again?

colethegamer123, 2020-10-08:
I made many custom settings inside the hack this is why I didnt wanted to rehack. The easiest method from user perspective would be a button what enables QR code scanning for 60 sec for example to scan the code from the app.

roleoroleo, 2020-10-08:
Swutitch off "Disable cloud" and pair the app.

colethegamer123, 2020-10-08:
I tried to enable, save, restart, start the app and try to pair. Nothing.

I tried to disable, save, restart, start the app and try to pair.
Nothing.

It should say something but nothing and it should scan tbe QR but nothing happens.

Any Idea or maybe only possible to unhack it and do reset everything? 
( would be bad for me )

Thank you :)

roleoroleo, 2020-10-09:
After switching off "Disable cloud" reset the cam using the back button.

colethegamer123, 2020-10-09:
After reseting all my settings will be lost what I changed on the files witth SSH right?

roleoroleo, 2020-10-09:
It depends on the files you are changed.
The reset operation restore only the default of the yi configuration.

colethegamer123, 2020-10-19:
In the new updates all things was added what I did manually, so I reseted the camera, added to yihome and online updated the hack firmware. Thank you

closed 2020-10-19 by colethegamer123