roleoroleo/yi-hack-Allwinner #155: YI home bricked after latest OTA update to 0.2.0 - warning!!

ferinelo, 2020-09-30:
This is to warn you. 
I have just clicked ota update from 0.1.9 to 0.2.0 and camera doesn't boot anymore.
I am trying to recover.

YI home 1080
YYS-2016
BFUS
fw 8.2.0

ferinelo, 2020-09-30:
Now I see https://github.com/roleoroleo/yi-hack-Allwinner#online-update-procedure :/
So maybe that is my fault - I haven't removed mp4 files.
I will give it a few more hours. Let's see...

ferinelo, 2020-09-30:
yep it works. My fault as written before.
Lessons learned: RTFM :-) :-P

closed 2020-09-30 by ferinelo

roleoroleo, 2020-09-30:
The original yi procedure checks the md5 sum of all files during update.
If there are many files, this takes a long time.

dentex, 2020-12-20:
I've just done the same error... upgrading to v0.2.2... but, to my surprise, all is OK now.

what I did:

- unplugged the camera
- deleted all video recordings
- copied manually the upgrade to the sdcard
- re-plugged the camera

And v0.2.2 running is OK