roleoroleo/yi-hack-Allwinner #156: MQTT motion ai_human not working propertly in 0.2.0

sermayoral, 2020-09-30:
Hi. I'm experiencing issues in v0.2.0 with MQTT.

i don't receive de 'motion_start' MQTT message anymore. I only receive the "motion_stop" message. 

The 'motion_files' MQTT messages always start in 1970:

```
{
"start":"1970-01-01T00:00:00+0000",
"end":"2020-09-30T16:08:26+0000",
"files":[ ]
}
```

AI Human MQTT messages don't work property. I understand it is in a experimental phase. Sometimes works, sometimes not.

I don't receive the baby crying messages anymore. 

Attached a test i have done. I hope
[mqtt.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/5306840/mqtt.zip)

Is anyone else experiencing this issues? 

Thanks

roleoroleo, 2020-10-01:
Please see here:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/125

I think it's no related to the hack, but you could try to unhack the cam and check if it's work.

sermayoral, 2020-10-01:
Thanks @roleoroleo. AI Human Detection disabled and everything works again :-)