roleoroleo/yi-hack-Allwinner #157: New YI Home 1080P AI+ BFUSY44GEE with FW 9.0.19.02_202007141512 supported?

RestOp, 2020-09-30:
Hello,
New YI Home 1080P AI+ BFUSY44GEE with FW 9.0.19.02_202007141512
I just got this cam https://www.amazon.es/gp/product/B01MR82NFZ
On the back side of Cam is BFUSY44GEE. Date on box 2020.09
I made a backup of FW. Here it is: https://drive.google.com/drive/folders/1ifFIDG4G0v-aIKrE_4kcy3elXd5Bsae3
Can not understand if it is supported or not. Could you please look and say if I need to send it back to amazon?

Thanks a lot!

danielk117, 2020-10-02:
Same here... BFUSY44*** and firmware 9.0.19.02_202007141512, bought on amazon germany.

lassieee, 2020-10-03:
me too. Got mine yesterday from amazon:
firmware 9.0.19.02_202007141512
DID: BFUS440PE...
SN: BY21GAEUW...
not sure if there's a hacked firmware for this model already.

[image]
[image]

danielk117, 2020-10-03:
Thanks for the teardown images 👍 
I tought these internal model names (like y20ga) came from the DID, but your board is marked as y29ga...

I think there isnt a hacked firmware for this device, because its just too new. But the QG2101A isnt new at all (see #70).
I hope @roleoroleo can tell if it belongs to MStar or Allwinner.

RestOp, 2020-10-03:
I hope also, in other way it must be send back to amazon 😌 I do need this cam if i can not intergrate it in HA 😀

danielk117, 2020-10-03:
> Guys, have you found any alternative to these cams in same price?
No...

roleoroleo, 2020-10-03:
I think it's an Allwinner model.
But the flash partitions are different.
home is a read only squashfs.
A new hack is needed.

RestOp, 2020-10-03:
> I think it's an Allwinner model.
> But the flash partitions are different.
> home is a read only squashfs.
> A new hack is needed.

It means we can forget and send it back or any chances in visible future?🤪

roleoroleo, 2020-10-03:
Moreover, the flash size is 8MB and not 16MB like the "current" Allwinner.
I think that the hack is possible but only using the external SD.

roleoroleo, 2020-10-03:
See here:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/105

Same hardware.

lassieee, 2020-10-03:
Sounds like a starting point. I wish I had the tools and skills to help in any way. If there is anything you think I could help with, please let me know!

danielk117, 2020-10-03:
> home is a read only squashfs.
> A new hack is needed.

You mean we need a root for this model? 
A a new hacks sounds like weeks or months, if anything is coming.

@RestOp Alternative in same price? I bought a 4pack of these cams on amazon germany just two week sago. All cams are 9FUS und flashable with this allwinner hack. But all cams have a disturbing noise coming trough the microphone (before and after hacking). So I wanted to send them back and bought new single ones (the new unknown model 😞).

roleoroleo, 2020-10-03:
Analyzing the file init.sh probably we could create an hack, like the Allwinner model.
But without a cam, I can't do it.

roleoroleo, 2020-10-04:
init.sh contains the same hack:
```
if [ -f "/tmp/sd/Factory/factory_test.sh" ]; then
        /tmp/sd/Factory/config.sh
        exit
fi
```
I could try to create a new telnet hack.

lassieee, 2020-10-04:
First of all thank you very much for your time and effort Roleo, it's much appreciated! 
I can imagine you really do need one of these new cams to create a successful hack, so perhaps we could:
A) see how many people are interested and split the cost so you can buy one of these cams to hack it, without any guarantees
B) I could send you my cam by mail and when you're done with it, you can return it
C) try to explain how the hacking procedure goes. I lack (proper) smd skills and jtag tools, but I'm a Linux engineer so I'm not completely clueless, and I can do some really simple soldering too.

roleoroleo, 2020-10-04:
I think I could start without a cam.
Then, if necessary, A would be the best.
In the meantime, if someone wants to try ...
[y21_hack.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5322868/y21_hack.tar.gz)

danielk117, 2020-10-04:
@roleoroleo I will give a try 😉

danielk117, 2020-10-04:
Factory was renamed to Factory.done, but no telnet

roleoroleo, 2020-10-04:
So, the script is executed but telnet doesn't start.

RestOp, 2020-10-04:
> So, the script is executed but telnet doesn't start.

What does it means? No way to hack?

roleoroleo, 2020-10-04:
No, no...
The hack s possible.

roleoroleo, 2020-10-04:
Ok. try this:
[y21_hack.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5323074/y21_hack.tar.gz)

lassieee, 2020-10-04:
Factory has been renamed to Factory.done, contains the file lower_half_init.sh which contains '/usr/sbin/telnetd &' (same as with your previous tar). Unfortunately nothing listens on port 23.
I was wondering if there is a telnetd in /usr/sbin, so i tried to ls -l /usr/sbin > /tmp/sd/Factory/sbin.txt but sbin.txt doesn't get created.

lassieee, 2020-10-04:
wait, scratch that, i reformatted my sd card and tried again, i got a telnet prompt now!

roleoroleo, 2020-10-04:
Ok.
Now the long part.

1. We need to create a new repo
2. Copy the Allwinner tree
3. Change all programs moving the binary from /home/yi-hack/ to /tmp/sd/yi-hack/
4. Check if h264grabber works or the framebuffer structure is changed
5. Check all compiled programs
6. ...

RestOp, 2020-10-04:
Unfortunatly, i'm not linux-friendly, but i'm study-friendly😎 so if any help from me needed, let me know!

lassieee, 2020-10-04:
Cool! If I can help in any way, please let me know :-)

danielk117, 2020-10-04:
Great 👍🏻 Will try it tonight 😊

roleoroleo, 2020-10-04:
I'm starting to create the new repo.
And effectiverly, without a cam it is very difficult. I can't ask you to try a new fw every 5 minutes.
Probably I need a cam...

RestOp, 2020-10-04:
> I'm starting to create the new repo.
> And effectiverly, without a cam it is very difficult. I can't ask you to try a new fw every 5 minutes.
> Probably I need a cam...

Understand, maybe u can try to make few different revisions and we will try them all and comment behavior?

roleoroleo, 2020-10-04:
This is the final (at the moment) version of the hack.
It overwrites /backup/init.sh with a "better" version that runs a shell script on the sd card at the end of the procedure.
If you install this hack you could start a new firmware on sd . If you remove the sd, the cam starts with the original fw.
[y21ga_hack.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5323799/y21ga_hack.tar.gz)

danielk117, 2020-10-04:
Works 👍 Im able to connect via telnet.

> If you install this hack you could start a new firmware on sd . If you remove the sd, the cam starts with the original fw.

What exactly do mean with "start a new firmware on sd"? 

Just unpack your latest home_y20gam.stage, copy all files from /home/yi-hack/ to /SDCARD/yi-hack/ and edit all paths in your bash scripts to the new path?

roleoroleo, 2020-10-04:
> What exactly do mean with "start a new firmware on sd"?

Wrong sentence.
It starts all additional programs.

> Just unpack your latest home_y20gam.stage, copy all files from /home/yi-hack/ to /SDCARD/yi-hack/ and edit all paths in your bash scripts to the new path?

It's not enough.
I have to change all the compilation/installation path.

danielk117, 2020-10-04:
Ok, tried it anyway 
[system.log](https://github.com/roleoroleo/yi-hack-Allwinner/files/5324119/system.log)

pep4n, 2020-10-05:
+1 for this. 

DID: BFUSY44...
SN_: BY21GAEU... 
FW: 9.0.19.02_202007141512

Let me know if I can test or support. 
Beer was just ordered ,) 
_Transaction ID: 8C490144TB577435B_

roleoroleo, 2020-10-05:
I'm working on it.
It will be a new repo.

roleoroleo, 2020-10-06:
I finished a first alpha release:
[y21ga_0.0.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5331644/y21ga_0.0.1.tar.gz)
Simply decompress it on your sd card.

Try it and give me a feedback.
In the meantime I will create the repo.

lassieee, 2020-10-06:
a little donation to get you started on that new cam: Transaction ID: 13M381272E579543T

roleoroleo, 2020-10-06:
> Beer was just ordered ,)
> _Transaction ID: 8C490144TB577435B_

> a little donation to get you started on that new cam: Transaction ID: 13M381272E579543T

@pep4n @lassieee 
Thank you!

lassieee, 2020-10-06:
FAT filesystem doesn't support symlinks :-(
The camera does need FAT, right?

roleoroleo, 2020-10-06:
Right.
I took this into consideration when I compiled busybox but there are probably other points...
For example yi-hack-v4 is a symlink, mqttv4 will not work (at the moment).

roleoroleo, 2020-10-06:
Removed all the symlinks:
[y21ga_0.0.2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5333848/y21ga_0.0.2.tar.gz)

lassieee, 2020-10-06:
sweet! I've been fiddling with this as well. I thought maybe we could add a script, or some lines to an existing script, that lets busybox on the camera make the symlinks, but it's been a busy day lol
I'll give this latest version a try and let you know

roleoroleo, 2020-10-06:
system.sh should start automatically.
You don't have to run it (if the procedure works).
Check if the processes are runnings: httpd, ftpdf, mqttv4, etc...

roleoroleo, 2020-10-06:
If you need the unhack procedure to remove the hack:
[unhack_y21ga.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5335076/unhack_y21ga.tar.gz)

danielk117, 2020-10-06:
> If you need the unhack procedure to remove the hack:
> [unhack_y21ga.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5335076/unhack_y21ga.tar.gz)

The LED remains orange. No matter if with or without SD card.

roleoroleo, 2020-10-06:
Very strange.

danielk117, 2020-10-06:
The camera does nothing except the orange LED.

roleoroleo, 2020-10-06:
Apply again the hack.

lassieee, 2020-10-06:
> system.sh should start automatically.
> You don't have to run it (if the procedure works).
> Check if the processes are runnings: httpd, ftpdf, mqttv4, etc...
> 
> EDIT
> 
> Found the problem...
> One more time around the block.
> [y21ga_0.0.2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5334974/y21ga_0.0.2.tar.gz)

I have no processes like httpd or ftpd running. I also can't find a binary for httpd. I've read that's part of busybox, but I'm unable to start if from a little test script. It says httpd not found.
I'm such a busybox n00b lol

danielk117, 2020-10-06:
> Apply again the hack.

Nothing happend. Factory isnt renamed anymore. I think its bricked.

roleoroleo, 2020-10-06:
> Nothing happend. Factory isnt renamed anymore. I think its bricked.

Did you clean the sd?
Is telnetd running?

roleoroleo, 2020-10-06:
> I have no processes like httpd or ftpd running. I also can't find a binary for httpd. I've read that's part of busybox, but I'm unable to start if from a little test script. It says httpd not found.
> I'm such a busybox n00b lol

I thought httpd was in the original busybox, but it's not there.
I will add it.

lassieee, 2020-10-06:
thanks. Can you also add hostname? that's missing as well

roleoroleo, 2020-10-06:
0.0.3: httpd and hostname
Please let me know if httpd starts.
[y21ga_0.0.3.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5335952/y21ga_0.0.3.tar.gz)

roleoroleo, 2020-10-06:
@danielk117 
This is the unbrick procedure for the older model:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam
I think it's pretty the same.
But you have to restore mtdblock4.

If you are able to connect the uart with a usb-serial adapter I can help you.

lassieee, 2020-10-07:
Ok, found it. In a previous version, we didn't have httpd so httpd couldn't start. Adding telnetd to /backup/init.sh DID work, so the check in Factory/config.sh skips copying newbackup/init.sh to /backup/init.sh because the hack is supposedly already in place.
httpd is running! 
[image]

roleoroleo, 2020-10-07:
Good!!!
This result is better than all my expectations.

> Ok, found it. In a previous version, we didn't have httpd so httpd couldn't start. Adding telnetd to /backup/init.sh DID work, so the check in Factory/config.sh skips copying newbackup/init.sh to /backup/init.sh because the hack is supposedly already in place.
> httpd is running! and so is ftpd, rstp..

Ok, new users shouldn't have this problem.

/tmp/sd/record:
Fixed removing the ln command from compile.www
Try to remove the symlink and it should work.
init.sh creates a bind mount.

Serial number:
The command (inside status.json cgi) is
SERIAL_NUMBER=$(dd bs=1 count=20 skip=592 if=/tmp/mmap.info 2>/dev/null | cut -c1-20)
dd and cut are present, probably the mmap.info file is different. Actually is offset=592, len=20
Check your mmap.info.

Memory:
free command is missing, I will add it in the next release.

Please check if rtsp stream is running and if the snapshot service works.
The 2 most complicated pieces.

danielk117, 2020-10-07:
> @danielk117
> This is the unbrick procedure for the older model:
> https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam
> I think it's pretty the same.
> But you have to restore mtdblock4.
> 
> If you are able to connect the uart with a usb-serial adapter I can help you.

I think this happend because i have the first hack installed (with the modified lower_half_init.sh and so on). 
Yes, I'm able to connect to uart (i own a TTL-interface and on @lassieee photos I see at least a RX, so there will probably be a marked TX too).
I'm still deciding whether to unscrew the device and destroy the warranty seal or just send the camera back (and hope that Amazon simply writes the device off as defective). 🙄

roleoroleo, 2020-10-07:
Ok, let me know if you need help.
I'm sorry about that.

pep4n, 2020-10-07:
> I think this happend because i have the first hack installed (with the modified lower_half_init.sh and so on).
> Yes, I'm able to connect to uart (i own a TTL-interface and on @lassieee photos I see [...]

I would recommend to send it back as DOA. 
Usually they don't check returns less than 40€.

[10 more replies]

closed 2020-11-21 by roleoroleo