roleoroleo/yi-hack-Allwinner #160: 0.2.0 memory leak?

ferinelo, 2020-10-06:
Hello,
I am facing an issue that seems to be a memory leak. On a regular basis free memory decreases from ~30MB to ~10MB and then system restarts (with delay).
Configuration:
* Yi 1080p Home BFUS | 8.2.0* | y20ga
* enabled: rtsp high +audio, swap, cloud, ntpd, mqtt, motion detection
* disabled: onvif, ssh, ftp, telnet, proxy, AI human detection, baby crying detection

By "restart with delay" I mean that when free memory is ~10MB or less I can still ping the device, but snapshot and rtsp don't work anymore. It will restart shortly after few tries of taking snapshot or streaming via rtsp.

[image]

Memory leak occurs when the camera is in idle, even without any active services or connections like rtsp, http or snapshot.

roleoroleo, 2020-10-06:
Check this issue:
https://github.com/roleoroleo/yi-hack-MStar/issues/149

Are you blocking internet access with a firewall rule?

ferinelo, 2020-10-06:
No, the camera has unrestricted access to the Internet. Cloud is enabled.

roleoroleo, 2020-11-21:
I fixed a memory leak in the onvif daemon.
I don't know if it solve your problem.
https://github.com/roleoroleo/yi-hack-Allwinner/commit/d39f23b4c5bc7b85273d3337a918c19fdfd0cc15

closed 2021-02-15 by roleoroleo