roleoroleo/yi-hack-Allwinner #163: ONVIF Home Assistant Connection Refused

Indianb0y016, 2020-10-15:
For whatever reason, the ONVIF integration in Home Assistant fails to connect to the camera, despite the ONVIF being enabled. Have tried with and without username and pass, still no dice. Ensured the port is correct. HA logs seems to suggest that the connection is refused, but Im not sure why.

roleoroleo, 2020-10-15:
Is the cam detected from discovery service?
Whel you start the integration wizard, does the cam appear in the list?

Indianb0y016, 2020-10-15:
When running the discovery, it does pick up the device as existing, however the name is "unknown". I would also add that the port is incorrect for the auto discovery. I DID set the port to 5000 in admin settings to see if that would work and unfortunately it doesnt.

roleoroleo, 2020-10-15:
If the discovery finds the device, don't worry if the name is unknown.
The discovery service should forward to the correct ip/port where onvif server is  listening.
There is no standard port for onvif, if you didn't change it, default for this cam is 80.

I'm using the cam with ha 0.116 and it works correctly.

Indianb0y016, 2020-10-15:
Strange. Perhaps maybe my asus merlin firmware has a firewall rule that I'm unaware of? Because even when it's detected, it still throws an "unknown error occured" problem. Logs continue to say that it's a connection refused problem.

roleoroleo, 2020-10-15:
Try with odm.

Indianb0y016, 2020-10-15:
Really appreciate your help mate. I know this problem looks elementary and Im so sorry for wasting your time. 
Using odm, it finds the camera, and its saying that there is no endpoint found. That must mean there is a setting problem with the camera. With onvif enabled, do I also need rtsp enabled? Because I thought Onvif was its own protocol that is independent of rtsp and such.

Indianb0y016, 2020-10-15:
Reset the camera to default settings from yi-hack and it had no trouble picking up the camera. Will close the issue now as it was my fault and a settings related problem. For anyone with similar difficulties, reset your camera to default y-hack settings and try again. I  foolishly disabled several settings to lessen the stress on the camera modules performance wise without realizing that it broke the onvif capabilities. Thank you Roleoroleo for assisting and I do apologize for wasting your time.

closed 2020-10-15 by Indianb0y016

roleoroleo, 2020-10-15:
No problem.
To answer your previous question: onvif is only a protocol descriptor so you need to enable rtsp.

Indianb0y016, 2020-10-15:
Noted. Thank you very much. Im still very new to all this so I appreciate the information! Have about four other yi cams with your legendary hack to work on. ;)