roleoroleo/yi-hack-Allwinner #173: Question about Camera Setting on always record?

chilman408, 2020-10-18:
Hello,

I'm not sure what this means under camera settings?

If enabled, video will be saved only when a motion is detected.
If disabled, video will be always saved.
(It takes effect only if recording is enabled)
--

Does this mean if I disable this setting, the camera will record 24/7?  Also where is the setting to enable recording?

roleoroleo, 2020-10-19:
> Does this mean if I disable this setting, the camera will record 24/7?

Yes, 24/7.

> Also where is the setting to enable recording?

First page.

chilman408, 2020-10-19:
Thx.  By first page, do you mean the "Recording wihtout Cloud" feature?  What if you have cloud enabled, is it still a 24/7 recording?

roleoroleo, 2020-10-20:
Yes, 24/7.

closed 2020-10-20 by chilman408